Duet3D/RepRapFirmware #202: short-to-ground frequently

gnydick, 2018-08-26:
I'm using this code on the M3D Promega. I've two Duet Maestro boards, both exhibit the same behavior. Two situations, but generate short-to-ground

1) extruder is experiencing a lot of resistance and triggers short-to-ground error. I'm guessing this is a misnomer and is not actually a short-to-ground

2) I've replaced my single Z motor with a dual-Z lead screw design and I frequently get short-to-ground when making movements, like homing Z. Again, I think this is a misnomer and more of a resistance detection, again, happens on both Maestros.

Is this tunable and can we get a more correct error message?

dc42, 2018-08-27:
Please ask firmware support questions on the forum at duet3d.com, not here.

closed 2018-08-27 by dc42

gnydick, 2018-08-29:
ok