Duet3D/RepRapFirmware #204: Include coordinates in the heightmap.csv file

caphector, 2018-09-01:
I have been troubleshooting calibration issues with my printer; I wanted to compare auto calibration results with the height map. Unfortunately, I don't see probe point coordinates in the heightmap.csv to assist in comparing probing points.

Could the XY coordinates of each probe be added to the heightmap.csv?

tekknix, 2018-12-06:
Hi caphector, 

even if this one is older, you will find the coordinates in the heightmap.csv if you hover your mouse over the Probepoints. Therefore you need to hover over every probed point.

Cheers
MoS-tekknix

dc42, 2018-12-06:
The first data line in the height map gives the start and end points and spacing for each axis. The probe point coordinates can be computed easily from those figures. The .csv format allows you to import the file into a spreadsheet, where you can do those computations.

closed 2020-10-20 by T3P3