Duet3D/RepRapFirmware #207: Duet Wifi does not reconnect to wifi on connection loss

KenMurphy, 2018-09-22:
I started a long running print last night. Shortly after starting the print I reset my WAP for an unrelated issue. The Duet Wifi did not reconnect to the AP. It has been printing fine for 12 hours so far, I am just unable to connect and view the web interface. I am also unable to ping the device.

WAP is WPA2. Duet is running 2.02RC2 and configured for DHCP. The router is configured to hand the duet a static IP.

dc42, 2018-10-02:
Currently the WiFi module attempts to reconnect twice when the connection to the AP is lost. In the next version of DuetWiFiServer I plan to change this to do continuous reconnect attempts.

cprezzi, 2018-10-02:
Wouldn't it be better to define the max count and delay between reconnect tries with a config param?

nemesis1782, 2019-04-07:
Any idea when this might get fixed? This often becomes a problem on long prints where color changes are required.

StephenRC, 2019-04-07:
https://duet3d.dozuki.com/Wiki/WiFi_disconnections_and_AJAX_timeout_errors

On Sun, 07 Apr 2019 00:41:30 -0700, nemesis1782
<notifications@github.com> had a flock of green cheek conures squawk
out:

>Any idea when this might get fixed? This often becomes a problem on long prints where color changes are required.

-- 

Stephen

. . . . File not found. Should I fake it? Y/N

---
This email has been checked for viruses by AVG.
https://www.avg.com

dc42, 2019-04-07:
This has already been fixed, at version 2.02.

stavros-k, 2019-04-07:
I was planning to add a button on the spare gpio pins to trigger a reconnect gcode, just in case of a failed connection.

closed 2020-10-22 by T3P3