Duet3D/RepRapFirmware #210: M558 F parameter only applied during approaching of probe

apphead, 2018-09-29:
I am using a BLTouch as Z-min and for bed leveling. Unfortunately I get bad repeatability due to the fact that the z-speeds for lowering and lifting are different. One is configured by the M558 F parameter (approaching the probe) while the other is defined by M203 Z parameter.

IMHO, this is a bug as the difference in speeds accounts for the bad repeatability (as I tested). **I propose that the M558 F parameter is applied to both directions in order to get a more consistent reading**, while the M558 T parameter is only applied to XY movements.
Lowering M203 Z is a possible workaround for bed.g auto-level, but not so for mesh grid compensation (G29). And sacrificing overall z-speed by lowering M203 Z in config.g to get more precision for G29 is just a last resort.

Tests performed on firmware 2.02RC2

dc42, 2018-09-29:
Pleaseruse the forum at https://forum.duet3d.com to make firmware suggestions, so that they will receive proper discussion.

closed 2018-09-30 by apphead

apphead, 2018-09-30:
Thanks for the hint not to create issues on github. I wondered but did not know that you prefer the forum. Is the category _General Discussions_ the right place to post bug reports / feature requests?

Posted in the [duet forum](https://forum.duet3d.com/topic/7100/m558-f-parameter-only-applied-during-approaching-of-probe/3)

dc42, 2018-10-02:
Feature requests should go in the firmware wishlist section. Reports of possible bugs can go in general Discussion unless there is a more appropriate section.