Duet3D/RepRapFirmware #212: M915 output gets clipped

TypQxQ, 2018-09-30:
I'm setting my M915 like this:

`M915 X Y S8 R1 F1`
`M915 E S5 R1 F1`
`M915 Z S7 R1 F1`

When I issue a M915 in "G-Code console" the output gets clipped:

> M915
> Driver 0: stall threshold 8, filter on, steps/sec 200, coolstep 0, action: log
> Driver 1: stall threshold 8, filter on, steps/sec 200, coolstep 0, action: log
> Driver 2: stall threshold 7, filter on, steps/sec 200, coolstep 0, action: log
> Driver 3: stall thr

I have my Duet connected to OctoPrint and issuing the same command gives me:

> Send: M915
> Recv: Driver 0: stall threshold 8, filter on, steps/sec 200, coolstep 0, action: log
> Recv: Driver 1: stall threshold 8, filter on, steps/sec 200, coolstep 0, action: log
> Recv: Driver 2: stall threshold 7, filter on, steps/sec 200, coolstep 0, action: log
> Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Send: M105
> Recv: Driver 3: stall throk T:130.0 /130.0 T0:130.0 /130.0 B:65.0 /65.0
> Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

I'm using:
Duet WiFi 1.02 or later
2.02RC2(RTOS) (2018-09-07b2)

Using the 2.01 gave e same results.

apphead, 2018-09-30:
@TypQxQ As dc42 told me yesterday, please post your findings on https://forum.duet3d.com/ since most people are looking there for further information / discussion etc. (I have no affiliation with duet, just trying to keep my main man dc42 focused on the development of the firmware ;-) )

> Configure long running commands or increase communication timeout
Seems more like an octoprint related issue. Have you tried to do that in octoprint?

TypQxQ, 2018-09-30:
Thankyou, found the answer on the forum:

> There is a limit on the response length for normal replies. You can use M915 D0, M915 D1 etc. to get status for individual drives.

That's why it crashes the Print function and everything in OctoPrint.

closed 2018-09-30 by TypQxQ

dc42, 2018-10-01:
The output is getting truncated, and I suspect that Octoprint is timing out waiting for the end-of-line character. I will change the command either to require that the D parameter is present, or to use a different output method that is suitable for long responses.