Duet3D/RepRapFirmware #219: Talk to ODrives via UART

tobbelobb, 2018-10-25:
Hi, I'm trying to use two ODrives with my DuetWifi V1.0. I want to communicate with them via UART.

Is is possible to configure the `URXD0/UTXD0` and `URXD1/UTXD1` pin pairs to get this communication up and running?

I don't mind having to unplug the wires connected to `URXD1/UTXD1` during firmware update. My attempts are in these commits:
https://github.com/dc42/RepRapFirmware/pull/186/commits/cfa62f7911d7453763d591290791262a57dda0c5
https://github.com/dc42/RepRapFirmware/pull/186/commits/d2b9485f90eb04041422520e380bef0b462e0b71

To my own surprise, I've had some limited success.
```
M569 Q0:1:115200
M114 S2
```
Did reply `2400` once, which is the true cpr of the ODrive. And my DuetWifi can still upload and update firmwares.

What should I think about to make sure I don't break the upload/update procedure by accident?

Where should I put the ODriveUART class?
It's now temporarily tucked into Platform.cpp and Platform.h...

Are there other things I should think about in order to have a reliable UART connection?

Best regards,

tobbelobb, 2018-10-26:
The ODrive could also use its UART protocol on its USB port, so if we could send a gcode to repurpose the DuetWifi's USB to UART, that would let us leave the SERIAL_WIFI_DEVICE (and the wifi's rx/tx pins) alone.

dc42, 2018-10-26:
Have you considered using one UART (probably the PanelDue port) and a multiplexer? That would allow expansion beyond 2 drives.

tobbelobb, 2018-10-28:
So I've looked into the multiplexer option again. Seems like I could buy a 74HC4052, stick it into a small breadboard, and basically build the multiplexer I need. I was looking so much forward to getting away from using small breadboards with the HP4 design.

I have also dived into the alternative of connecting the second ODrive to the UTXD1/URXD1 pins. In DuetWifi's schematics I've found that there are resistors between the ESPs TX/RX and the SAMs RX/TX. In my understanding, this should prevent the ESP and the ODrive from burning each other (given that they are connected to different sides of the resistors). If the ODrive were to send HIGH while ESP tried to send LOW or vice versa, only a ~mA current would get through.

It should even be possible to use the ESP's RX/TX during programming without unplugging the wires that go to the ODrive. However, the ODrive could interrupt (or garble) the ESPs UART signal, because the ODrive would not communicate through resistors, so it's signal would take precedence over the ESP's signal.

In the unlikely event that ODrive used its RX/TX lines during RepRapFirmware upload, we could just try again, eg re-upload, couldn't we?

I'm about to try using UTXD1/URXD1 as if they were any other pair of UART pins. If you're reading this and think that I'm about to brick my beloved DuetWifi boart, please tell me. I haven't done stuff like this before.

dc42, 2023-04-11:
Closing this because communicating with ODrives via the secondary CAN bus on the MB6HC and MB6XD has been implemented.

closed 2023-04-11 by dc42

k-kuehn, 2024-04-25:
Where is the CAN communication with Odrive documented?

T3P3, 2024-04-25:
https://docs.duet3d.com/en/User_manual/Reference/Gcodes#m5693-read-motor-driver-encoder-via-secondary-can-bus
https://docs.duet3d.com/en/User_manual/Reference/Gcodes#m5694-set-motor-driver-torque-mode
https://docs.duet3d.com/en/User_manual/Reference/Gcodes#m5698-read-motor-force-via-secondary-can-bus