Duet3D/RepRapFirmware #221: Scara Min Radius

NicholasSeward, 2018-10-30:
@dc42 
I was asked by Danny Defelici to help out with this.  For SCARA machines, you want to avoid running the machine into itself.  For HELIOS, I have implemented a nice mapping of points that would cause crashes to the nearest reachable point.  That would be nice in the general case but for Danny's SCARA I was thinking we could just implement a simple min radius.  

Example:
minRadius=100;
(30,40) is requested but machine actually goes to (60,80) 

Do you think it is appropriate to modify CalculateThetaAndPsi where it give an erroneous Theta and Psi and just let's the machine think it is at an unreachable x,y?  I have done that and even if you try to print into that region it still prints fine but the clipped part gets smashed a bit.

Basic psuedocode to insert between line 41 and 42

d=hypot(x,y)
if d<minRadius:
   x=x*minRadius/d
   y=y*minRadius/d

The same can be done for maxRadius so that a print that just barely goes out won't cause a fail and will just clip that part off.

if d>maxRadius:
   x=x*maxRadius/d
   y=y*maxRadius/d

What do you think?

NicholasSeward, 2018-11-16:
I see that there is a function IsReachable.  Does this get queried before the machine tries to move to a point.  If so...all the logic is in place.  We would just need to give a way to modify minRadius from gcode.

NicholasSeward, 2019-01-22:
The problem as I understand it...

Psi is homed by having the distal arm fold in on the proximal arm.  There is a sensor on the proximal arm that is triggered when this happens.  The nozzle at this point is located above a housing for the harmonic drive.  If you return to this psi location during the first few layers of a print then there will be a collision.

The firmware solution could be to home high above the harmonic drive housing and then move the nozzle out to an enforced safe zone.  Right now min_radius is calculated.  If it could be set then this would be one possible solution.  Do you see any other firmware or gcode solutions we could use to prevent a machine crash?

dc42, 2019-01-23:
I wasn't aware of the protruding harmonic drive. How easy would it be for Danny to change the position of the distal homing switch so as to prevent that collision even when the head is very low?

NicholasSeward, 2019-01-23:
Haha.  It is possible to change but not optimal.  For the best homing the switch needs to be far from a pivot.  To have the sensor far from the pivot but in a location to protect the housing would require a long protrusion from one of the arms.  Additionally, he has shipped some printers already and is looking at ways to make them safer.

I was thinking of a solution that would keep min_radius calculated unless it was explicitly set.  This would maintain backwards compatibility.  Would that solution work within the project standards?

dc42, 2019-01-23:
Yes that would be possible. There is currently no R parameter, so I could allow Raaa:bbb to set the min:max radius, if the supplied values are higher:lower than the calculated values.

NicholasSeward, 2019-01-23:
Sweet.  Would you like me to do a first stab at the code to save you some time?  I don't have my environment setup so I won't be for sure that it compiles but I usually get pretty dang close.

dc42, 2019-01-23:
I have already implemented it. Code will be in next commit.

dannydefe, 2019-01-24:
David, Nicholas
  thanks for working on this issue.
Danny

closed 2020-10-22 by T3P3