Duet3D/RepRapFirmware #226: CoreXY kinematics as used in Markforged machine.

leolongvu, 2018-11-14:
Hello

Is it possible to add a support for the Markforged kinematics as described in the picture? I upgraded my printer to this type of kinematics because the planar belt path makes everthing a little bit tidier.

[image]

dc42, 2018-11-15:
See https://github.com/dc42/RepRapFirmware/blob/dev/AddingNewKinematics.md

StephenRC, 2018-11-15:
Isn't that a variation of a Cartesian printer?

On Wed, 14 Nov 2018 21:19:33 +0000 (UTC), leolongvu
<notifications@github.com> had a flock of green cheek conures squawk
out:

>Hello
>
>Is it possible to add a support for the Markforged kinematics as described in the picture? I upgraded my printer to this type of kinematics because the planar belt path makes everthing a little bit tidier.
>
>[image]

-- 

Stephen

CONGRESS.SYS Corrupted: Re-boot Washington D.C. (Y/N)

---
This email has been checked for viruses by AVG.
https://www.avg.com

leolongvu, 2018-11-15:
@StephenRC No this is a variation of the CoreXY kinematics, the only difference is that the Y axis is only driven by the Y motor.

I am having a little bit of trouble building the firmware binary to test the kinematics code, will submit the new kinematics once I tested them out.

StephenRC, 2018-11-15:
One motor moves the Y.  One Motor Moves the X.  Just like a Cartesian.
The X just has a more complicated belt path.

On Thu, 15 Nov 2018 08:09:19 -0800, leolongvu
<notifications@github.com> had a flock of green cheek conures squawk
out:

>@StephenRC No this is a variation of the CoreXY kinematics, the only difference is that the Y axis is only driven by the Y motor.
>
>I am having a little bit of trouble building the firmware binary to test the kinematics code, will submit the new kinematics once I tested them out.

-- 

Stephen

CONGRESS.SYS Corrupted: Re-boot Washington D.C. (Y/N)

---
This email has been checked for viruses by AVG.
https://www.avg.com

mveinot, 2018-11-15:
I don't think that's quite true. If you hold X stepper still and move Y stepper, the X axis will still move, because the relative belt path lengths will change. X will have to move in some amount to compensate for this.

leolongvu, 2018-11-15:
@mveinot That's totally correct.

By the way the reason I am not able to test the kinematics code is that there is some trouble with compiling the CoreNG project, specifically this error concerning the Wire arduino library:

`Building file: ../cores/arduino/Print.cpp
Invoking: Cross G++ Compiler
arm-none-eabi-g++ -D__SAM3X8E__ -DUDD_ENABLE -DUDD_NO_SLEEP_MGR -I"C:\Eclipse\Firmware\CoreNG\cores\arduino" -I"C:\Eclipse\Firmware\CoreNG\asf" -I"C:\Eclipse\Firmware\CoreNG\asf\sam\utils" -I"C:\Eclipse\Firmware\CoreNG\asf\sam\utils\header_files" -I"C:\Eclipse\Firmware\CoreNG\asf\sam\utils\preprocessor" -I"C:\Eclipse\Firmware\CoreNG\asf\sam\utils\cmsis\sam3x\include" -I"C:\Eclipse\Firmware\CoreNG\asf\sam\drivers" -I"C:\Eclipse\Firmware\CoreNG\asf\common\utils" -I"C:\Eclipse\Firmware\CoreNG\asf\common\services\clock" -I"C:\Eclipse\Firmware\CoreNG\asf\common\services\ioport" -I"C:\Eclipse\Firmware\CoreNG\asf\common\services\usb" -I"C:\Eclipse\Firmware\CoreNG\asf\common\services\usb\udc" -I"C:\Eclipse\Firmware\CoreNG\asf\common\services\usb\class\cdc" -I"C:\Eclipse\Firmware\CoreNG\asf\common\services\usb\class\cdc\device" -I"C:\Eclipse\Firmware\CoreNG\asf\thirdparty\CMSIS\Include" -I"C:\Eclipse\Firmware\CoreNG\libraries\Storage" -I"C:\Eclipse\Firmware\CoreNG\variants\duet" -O2 -Wall -c -std=gnu++17 -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -nostdlib -Wdouble-promotion -fsingle-precision-constant "-Wa,-ahl=Print.s" -MMD -MP -MF"cores/arduino/Print.d" -MT"cores/arduino/Print.o" -o "cores/arduino/Print.o" "../cores/arduino/Print.cpp"
../libraries/Wire/Wire.cpp:247:1: error: expected declaration before '}' token
 }
 ^
Finished building: ../libraries/Storage/sd_mmc_mem.c
Finished building: ../libraries/RTCDue/RTCDue.cpp
 
 
Building file: ../cores/arduino/Reset.cpp
Building file: ../cores/arduino/RingBuffer.cpp
libraries/Wire/subdir.mk:18: recipe for target 'libraries/Wire/Wire.o' failed
make: *** [libraries/Wire/Wire.o] Error 1
Invoking: Cross G++ Compiler
make: *** Waiting for unfinished jobs....`

I am not entitly sure about the reason behind it, maybe the "make" command, maybe the Arduino itself.

StephenRC, 2018-11-15:
Dang, didn't see that the x and y move the head. The x has to move
whenever the y moves.

On Thu, 15 Nov 2018 08:56:18 -0800, Mark Veinot
<notifications@github.com> had a flock of green cheek conures squawk
out:

>I don't think that's quite true. If you hold X stepper still and move Y stepper, the X axis will still move, because the relative belt path lengths will change. X will have to move in some amount to compensate for this.

-- 

Stephen

CONGRESS.SYS Corrupted: Re-boot Washington D.C. (Y/N)

---
This email has been checked for viruses by AVG.
https://www.avg.com

T3P3, 2020-10-22:
I beleive this is now possible using the Kinematics matrix in RRF from 2.03 and later:
https://duet3d.dozuki.com/Wiki/Gcode#Section_Parameters_for_Cartesian_CoreXY_CoreXZ_CoreXYU_CoreXYUV_MarkForged_kinematics_RRF_2_03_and_later_only

Please start a forum discussion if that is not the case.

closed 2020-10-22 by T3P3