Duet3D/RepRapFirmware #228: Function "delay" and "micros" are not declared

leolongvu, 2018-11-16:
Hello

I am trying to test the code for a custom kinematics, however I am unable to compile the RepRapFirmware project in Eclipse. I can compile CoreNG, FreeRTOS and RFFLibraries no problem.

The errors have been here on this forum once but no one gave an answer and the thread was closed. Eclipse throws the followings:

../src/Storage/FileStore.cpp:295:18: error: 'micros' was not declared in this scope

../src/Storage/MassStorage.cpp:83:3: error: 'delay' was not declared in this scope

src/Storage/subdir.mk:24: recipe for target 'src/Storage/FileStore.o' failed
make: *** [src/Storage/FileStore.o] Error 1
make: *** Waiting for unfinished jobs....

make: *** [src/Storage/MassStorage.o] Error 1
src/Storage/subdir.mk:24: recipe for target 'src/Storage/MassStorage.o' failed

I guess the later two are the results of the first two. I messed with the files the whole day but couldnt figure out a resolution. Anyone got any ideas? Cannot test my custom kinematics code without compiling it.

dc42, 2018-11-16:
It sounds to me that you are using incompatible branches of RRF and other projects. https://github.com/dc42/RepRapFirmware/blob/dev/BuildInstructions.md

leolongvu, 2018-11-16:
I am mkaing myself sure that I have downloaded the correct firmware branch. RRF and CoreNG are master branchs, Reprapfirmware dev-2 branch, the other two are also master branch.

I have messed around with the code and got it up to a point where it says:

"C:\Eclipse\Firmware\RepRapFirmware\src/RepRapFirmware.h:38:10: fatal error: General/SafeStrtod.h : No such file or directory"

I know that SafeStrtod.h is belong to RRFLibraries, but I cant get Eclipse to detect the file. Path and Symbols are updated, references are updated. I guess something is wrong with Eclipse.

leolongvu, 2018-11-16:
@dc42 
Ok I got what the error is. With the default configuration the Firmware cannot be compiled. With the new Duet Wifi configurations should be SAM4E, SAM4E_ROTS and Duet2_RTOS. I think you should put this one in the build instruction that reminds people to choose the correct configuration.

closed 2018-11-16 by leolongvu