Duet3D/RepRapFirmware #230: RepRapFirmware branch v2-dev corrupted.

leolongvu, 2018-11-17:
Hello

After a lots of tries I have been able to rebuild the most recent firmware as described in the BuildInstructions.md. I then tried to upload the firmware to the printer to check if it works before implementing my custom kinematics code. Unfortunately as the firmware updated the web interface shown the following notification:

"Could not establish a connection to the Duet firmware! Please check your settings and try again."

I then tried to connect the board to the PC via serial using Prontface and YAT. The connection is just frozen with Prontface, meanwhile I am able to connect to YAT but unable to get any feedback from the printer (e.g M115). I had to do a hard erase and reupload the precompiled firmware to get the connection back. 
I also tried to recompile the order version of the firmware but the mismatch branches of other projects just make it impossoble to do so, I always get random errors. Is there a way to check which part of the firmware is corrupted?

dc42, 2018-11-17:
AFAIK the current committed sources (v2-dev for RRF, main for everything else) work together; however I expect to commit changes to all 4 projects today, and this set definitely works.

leolongvu, 2018-11-17:
It works after the update, thank you very much.

closed 2018-11-17 by leolongvu