Duet3D/RepRapFirmware #231: PWM Multiplier for 12v components

svenyonson, 2018-11-17:
Feature request: I am using 24v to power my Duet 2 Wifi, (upgraded my Ender 3) and my printer is now SUPER quiet, thanks to the TMC2660s. Now the loudest part of the printer is the fans. The quietest fans I can find (noctua) are only made in 12V versions. Instead of adding circuitry to accomodate these, I thought it would be useful to have a multiplier that could be applied to individual PWM circuits (typically 50%).

zafuquazar, 2018-11-17:
I too, am In the same boat. I have not tried it yet, but can’t you just set the PWM to 50%? I’m not sure adding a multiplier to the code is necessary. 

By the way, a few have described an excellent way to solve our problem. I have not implemented yet, but here’s the jist: 
The power rail for the fans is on the center pin of the three pins where you put the jumper that controls the fan voltage. The outer pin on one side is supply voltage (24V for you and I) and the other pin is from the 5V regulator. You can remove the jumper and wire a 12V regulator to that center pin. Now the entire fan rail is powered at 12V. The PWM is on the ground, so this does not interfere with PWM.

svenyonson, 2018-11-17:
Excellent! Thanks for the reply. I will try this. Honestly, I hadn't even noticed the fan voltage jumpers.

zafuquazar, 2018-11-17:
Yeah, I was super excited when someone told me about this solution :)
I haven’t done it myself yet. I assume a voltage regulator using the printer power supply is Bette than a separate 12V supply, so there’s no ground loops involved in the PWM stream

Good luck!

svenyonson, 2018-11-17:
I made a simple harness using dupont connectors - the 5 pin connector spans the fan voltage select and the internal 5V enable jumper blocks, making it easy to get to GND.
[image]

zafuquazar, 2018-11-19:
Hey, that is beautiful! Very clever!

Gerald3, 2020-09-26:
Awesome, for those that don't know much about electrical engineering talk / much at all - Could I ask what the unit on the right side of the custom wiring harness is? is it a buck converted from 24 to 12? The ones I have are huge (and I've stupidly blown out all fans many times by not knowing why I couldn't just send the output of the fans in 24V to the converted and have it control the 12V .. fan or light, etc.  Apparently, you can't do that so  I have to use a MOSFET then that conversion means... AKA A LOT OF STUFF BEING ADDED.    -  I just wanted to know if someone could let me know about the electronics info here / like a shopping list to vastly simplify my builds??   This does convert a 24V system to having ALL FAN Outputs / controllable and always on all running at 12V correct?  Thanks guys!

svenyonson, 2020-09-26:
It is just a L7812 12 volt voltage regulator.

svenyonson, 2020-09-26:
https://www.amazon.com/MCIGICM-Voltage-Regulator-Linear-Positive/dp/B07D9Z7TDS/ref=sr_1_3?dchild=1&keywords=L7812&qid=1601134333&sr=8-3

svenyonson, 2020-09-26:
This only affects fans connected to V-FAN

dc42, 2020-09-26:
Please discuss this sort of thing on the Duet3D forum. It has nothing to do with RepRapFirmware.

closed 2020-09-26 by dc42