Duet3D/RepRapFirmware #234: HTTP responses doesn't conform to HTTP spec

martinj, 2018-12-06:
Was trying to interface with the HTTP api using a micro controller, but i just couldn't manage to handle the responses with the http library I used. After some debugging i noticed that Duet doesn't use CRLF after each header but only LF. 

Skimmed trough the code and I think at least one part of the culprit is here where only LF (`\n`) is used instead of CRLF (`\r\n`) https://github.com/dc42/RepRapFirmware/blob/6b835cb3c8e9aa10113ae750151d9f9065116cd4/src/Duet/Webserver.cpp#L679-L682

I'm not well read into the specifics https://www.w3.org/Protocols/rfc2616/rfc2616.txt but I do think that headers isn't the only part that requires termination using CRLF.

dc42, 2018-12-06:
A similar issue was raised in relation to the build of RepRapFirmware for the later Duets, and I fixed it in the latest build (2.0RC5). However, the older Duets use different webserver code, and I no longer maintain that verison.

martinj, 2018-12-06:
Ok. I'll try with latest version. Will reopen if the issue remains in latest version.

closed 2018-12-06 by martinj

dc42, 2018-12-08:
I've now fixed this issue in the source code for the webserver for the legacy Duets; however although RRF can still be built for the legacy Duets, it is not guaranteed to run because of other changes.