Duet3D/RepRapFirmware #240: Lots of Dual Z endstop bugs

bobbyofna, 2018-12-19:
Hello, so I am currently working on my custom designed Mini CoreXY 3D printer and am currently in the final stage of trying to configure reprapfirmware.  In doing so, I have ran into a number of bugs trying to get the dual Z axis endstops to work properly.  I've been messing with it all day today and yesterday, and after reading every forum relating to reprapfirmware and dual Z endstops, and trying tons of different things, here are all of the problems I can remember.  Keep in mind for everything I list here, I have likely tried in 2.02RC4, RC5, and RC6.  However, RC6 seems to be broken for my configuration as no axis will move whatsoever (not X, Y, Z, U, nor E0).  Homing doesn't move anything in RC6, nor do any G1 commands, again for all axis'.  I did try out some changes that made a difference in RC4, and RC5 on RC6, but not every change.  It was always the same result for RC6 every time, that being that nothing would move at all.

The first problem I had was that the two Z motors (Z and U) were moving at different speeds when homing.  When homing, Z moved as it should, but U just was barely moving, as in I could feel the slow individual steps when lightly holding the leadscrew coupler.  Moving the Z axis (Z and U together) with G1 Zxx Fxxxx worked perfectly fine whether I used the web interface or the paneldue, but it totally misbehaved when homing.  This problem was pretty much the same as far as I can remember for both RC4 and RC5.  To move the Z axis together with G1 however, I usually had to first send M564 S0 H0 so I did not have to home first.

I finally found a temporary fix for everything but it's not ideal, and it results in more bugs/glitches.  This fix only works on RC4, and consists of these changes...
1) (config.g) Do not set a Z and U axis minimum
2) (homez.g) Added M584 P4 as the first line
3) (homez.g) Changed M584 Z2 to M584 Z2 U3 P4
4) (homez.g) Changed M584 Z2:3 to M584 Z2:3 U3 P3
5) (homez.g) Set the location afterwards, have the Z axis do one more unnecessary move, and set the location again with G92

For fix #1, if you DO set an axis minimum for X and U in config.g and try homing the Z axis, it will home synchronized, but it overrides the G92 value set for the Z location in homez.g, and sets the location of Z after homing to whatever the Z axis minimum value is as defined in config.g.  The Z axis will not move after this, unless you send the command M564 S0 H0.  However, after doing which, if you try and move the Z axis it will only move between your Z axis minimum and 0, and the direction of the Z axis (while synced) reverses after you move it once.

With fixes #1, through #4, it will home properly, set the Z location as defined by G92 in homez.g as its supposed to, and let you move the Z axis afterwards.  However, the Z home button on the paneldue stays in the blue purple clicked color (despite it still being clickable), and the next Z move made after M584 Z2:3 U3 P3, moves extremely slowly.  After that slow move, the Z axis moves perfectly fine and synchronized (the button still looks clicked), hence fix #5.  Re homing does the same thing, no changes in behavior if I just keep homing Z without rebooting the Duet Wifi.

I've have tried many other things that didn't seem to make any difference, for example...
-- Messing around with things in the config like doing Zxx Uxx, or Zxx:xx, or Zxx:xx Uxx.
-- Trying many combinations of those for just about every command in my config that references the Z axis.
-- Sometimes using M564 S0 H0 in conjunction with a lot of those changes.

That's all of the problems and details I can think of.  I feel like there is more and will report if there are.  (Also, despite having this issue with the Dual Z endstops, I am already loving my Duet even though I haven't printed anything yet).  You can find my config.g and homez.g at the bottom as well as technical details about my build, and things I've tried as suggested in some other forums.  I am on break for 3 more weeks before the next semester and can be the guinea pig for testing if need be.

The printer...
https://www.thingiverse.com/thing:2806723

Some other general notes...
-- Using the newest Duet 2 Wifi 1.04
-- Paneldue v2.0 7" (current firmware version is 1.22.1)
-- Wifi server version is 1.21
-- Web interface version is 1.22.5
-- I have not tried anything older than 2.02RC4
-- BLTouch for bed leveling
-- Dual Z max endstops
----- M119 shows that the Z and U max endstops are working fine
-- All motors seem to generally work with my fix (haven't done a lot of testing yet, still need to fine tune the sensorless homing and try bed leveling)
-- Heaters work perfectly fine even with my fix

`;************************* CONFIG FOR DUET 2 WIFI W/V2.02 FIRMWARE *************************;

; * ****************** * GENERAL * ***** * GENERAL * ***** * GENERAL * ****************** * ;
    M111 S0                         		; Debug off
    M550 P"Mini CoreXY"				; Machine name and Netbios name
    M551 PMelonPr1sm           			; Machine password (used for FTP)
    M555 P2					; Set output to look like Marlin
    M552 S1 P"Musical8"				; Turn network on
    M575 P1 B57600 S1				; Comms parameters for PanelDue
    G21						; Work in millimeters

; * ********* * MOTOR  CONFIG * ***** * MOTOR  CONFIG * ***** * MOTOR  CONFIG * ********* * ;
    M667 S1 					; Switch to CoreXY mode
    M569 P0 S1					; Drive 0 goes forwards
    M569 P1 S1					; Drive 1 goes forwards
    M569 P2 S1					; Drive 2 goes forwards
    M569 P3 S1					; Drive 3 goes forwards
    M569 P4 S0					; Drive 4 goes forwards
    M584 X0 Y1 Z2:3 U3 E4 P3			; Defines which driver is which motor (w/dual Z)
    M671 X-97.5:231.5 Y80:80 S5			; Define where the 2 lead screws are (max of 5mm correction)
    M574 X1 Y1 S3				; XY axis' use sensorless homing at minimum
    M574 Z2 U2 S0				; Z axis uses endstops (normally open) at maximum
    M915 P0:1 S8 F1 H400 R0			; Sensorless homing
    M915 P2:3 S8 F1 H200 R1			; Stall detection (rehome.g when stall is detected)

; * ********* * AXIS SETTINGS * ***** * AXIS SETTINGS * ***** * AXIS SETTINGS * ********* * ;
    M350 X16 Y16 Z16:16 U16 E16 I1		; Set 16x microstepping with interpolation
    M92 X160 Y160 Z400:400 U400 E476.5		; Set axis steps/mm
    M906 X1700 Y1700 Z1700:1700 U1700 E1700 I60	; Set motor currents (mA) and idle current (I)
    M913 X50 Y50 Z50:50 U50 E50 		; Set motors to 50% of their normal current
    M201 X2500 Y2500 Z100:100 U100 E10000	; Accelerations (mm/s^2)
    M203 X20000 Y20000 Z1200:1200 U1200 E10000	; Maximum speeds (mm/min)
    M566 X600 Y600 Z20:20 U20 E600		; Maximum jerk mm/minute
[... 25 lines cut ...]

`;*****homez.g*****;
M584 P4			; TEMPORARY FIX
G91 			; Relative positioning
M584 Z2 U3 P4 		; Split Z into 2 (Z+U)

G1 Z250 U250 F600 S1	; Move to endstops
G1 Z-10 U-10 F600 S2 	; Move away from endstops

M584 Z2:3 U3 P3		; Join U to Z again

G92 Z125		; Set location
G90 			; Absolute positioning
G91 			; Relative positioning
G1 Z-5 F600		; Move that is always slow despite what speed you set this command to
G92 Z125		; Set location again
G90 			; Absolute positioning`

Forums of people with similar issues and solutions I have tried...
https://duet3d.dozuki.com/Guide/Independent+Z+motors+and+endstop+switches/18?lang=en
https://forum.duet3d.com/topic/7755/drive-3-is-slower-as-drive-2-at-homing-in-z-solved/32
https://github.com/dc42/RepRapFirmware/issues/91
https://forum.duet3d.com/topic/2688/2x-z-axis-motors-home-together-then-sync-invididually-need-input/9
https://forum.duet3d.com/topic/6974/problem-with-3-independent-z-axis-motors-and-endstops/21
https://forum.duet3d.com/topic/3938/need-help-with-m584-to-sync-3-z-motors/9

bobbyofna, 2018-12-19:
Update: RC6 now kinda works, and better so than RC4 and RC5.  I am not exactly sure what I did yet that made it work but the main changes I have made since posting this are some changes to the sensorless homing stuff in the config.g, no axis minimum for all axis' in config.g, and you apparently have to home Z first when it boots, otherwise it doesn't let you move anything else.

bobbyofna, 2018-12-19:
Update: I fixed most of the issues I was encountering (also I am an idiot, I realized the axis home button stays blue if it's been homed first).  The issues I was having were primarily stemming from the M208 command.  For one, I apparently had a redundant M208 command In my config.g file.  The other thing I noticed is in the duet gcode documentation, the S parameter values seem to be backwards.  Adding Sx of the opposite target axis end (min or max) to the M208 command fixed a number of issues, even more so if you separate the min M208 dedinition into two, XY min and Z min.  The only issues left are the slow Z movement after homing the Z axis, and that RC6 won't let me do anything until I home Z first.  For right now I just put G28 Z at the end of my config.g file.  That seems to work.  The single slow movement I can't seem to fix.  It moves whatever distance I tell it to but ignores the speed I set.  The speed seems to be dependent on the distance traveled and time, meaning it might always take ~5 seconds to move no matter what distance you set it to.  Haven't confirmed that though.  Will try it tomorrow and provide more details as well.

dc42, 2018-12-19:
Please use the forum for support, see the readme file of this project.

closed 2018-12-19 by dc42