Duet3D/RepRapFirmware #242: reprapFirmware 2.0 vs marlin 2.0. Clash of the Titans

vladimirRUS007, 2018-12-21:
Marlin 2.0 and reprapFirmware.
Board used RURAMPS, Arduino DUE.
Is the firmware stable and productive for the Arduino DUE compared to the 32 bit reprapFirmware.
Interested in functionality and performance. And how will this affect print quality?

dc42, 2018-12-22:
Hi Vladimir, are you the author of the RURAMPS fork of RRF?

The latest version of RRF is 2.02RC7 provisional, which I have just committed. It will be released later today subject to satisfactory testing. The most recent commits should only affect the web interface and CNC machines running RRF.

badvision, 2020-01-22:
It's probably ok to close this for now since it's well-answered, no?

dc42, 2020-01-22:
Yes.

closed 2020-01-22 by dc42