Duet3D/RepRapFirmware #244: Web interface - Endless loop in files and macros

kafledelius, 2018-12-25:
Hello.

After upgrading to the latest firmwares:

Firmware Version: | 1.20.1 (2018-01-08)
Web-Server: 1.20.1
Web interface: 1.22.6 (Tried downgrading)

I'm getting endless loops in the Macros and the Gcode files section.

Is this a known issue ?

Jester3rdDimention, 2019-01-31:
Yeah, I am running into the issue. I have tried to remove the macros and the Gcode files and re update but it still is happening, if i go back to 1.20 then it stops happening but any version high i get this issue.

[image]
[image]
[image]

dc42, 2019-11-08:
This issue is caused by using incompatible versions of RepRapFirmware and DuetWebControl.

closed 2019-11-08 by dc42