Duet3D/RepRapFirmware #265: [FR] Allow different font sizes and other symbols for 12864 on Duet Maestro

shoreliner, 2019-02-16:
Due to limited screen space, it would be ideal to be able to change font size to even lower and also include some basic symbology in the character set if this is possible.  As of now, I could produce much less cluttered main menus with smaller fonts for the Crane Series.

Thanks,

David
M3D Team

dc42, 2019-02-16:
The firmware structure allows for additional font sizes. So please go ahead and generate one or more smaller fonts. I used Glcd Font Creator to generate the original fonts. For me to accept your pull request, you must convert all Unicode code points up to at least U+0xFF (U+0x17F is better), and start from an open source font (e.g. Liberation Sans) to avoid copyright issues. The smaller the font size, the more hand-editing it will need to be acceptable.

shoreliner, 2019-02-16:
Will Do -- Thanks Dave. I've also begun to accumulate a large number of original "Glyphs"/Icons for use on the 12864, I'll make those available somehow as well if you'd like.

I will get into Font creation ASAP.

David
M3D Team

shoreliner, 2019-02-18:
@dc42 I've set to work on this.  Can you clarify for me the font size (X by X) that are currently being used as Font "0" and "1" in the Maestro firmware?  I would just like to have a basis to work from.

Cheers

David

dc42, 2019-02-18:
The font sizes are as the filenames, so 7x11 pixels and 11x14 pixels. The first number is the width, the second is the height.

shoreliner, 2019-02-18:
Thank you sir.  I was considering going with a Monospaced font but... perhaps that isn't the best idea.  I'm really just trying to get a bit more LCD screen real estate while ensuring things aren't too cluttered.

Regarding font licensing: OFL(Open Font License) would be acceptable in this case if i'm not mistaken.

dc42, 2019-02-18:
Monospaced fonts look horrible and takeup more width.

shoreliner, 2019-02-18:
Thanks for your help with this.  Would you prefer I close the issue and submit via PR when completed?

shoreliner, 2019-02-18:
Will submit as PR when complete.

closed 2019-02-18 by shoreliner