Duet3D/RepRapFirmware #266: G10 changes Z=0 for following prints if not followed by unretract (G11)

Armadillo1, 2019-02-17:
As discussed in https://forum.duet3d.com/topic/9146/mid-air-prints-unless-i-restart/13 I believe I found a solution to varying Z height between prints.

The problem I had was increased Z height if i paused and cancelled a print and wanted to start a new one. It would start printing too high.

Apparently if you have G10 in your pause.g (or, I suppose anywhere in your Gcode) and have Z hop defined in M207, after cancelling the print (or, I'd guess, finishing it without unretracting), after starting a new print, z=0 becomes the value of the Z hop height in M207, because it zhops, the print gets cancelled or finishes and never actually zhops down to previous height.

I tested this by uncommenting G10 from my pause.g and the problem disappeared. I also tried setting M207 Z(random number) and after starting a new print, Z=0 is at a height of (random number) and actually the nozzle can be jogged down into negative numbers after stopping the print...and it goes right to the negative height of M207 Z value.

This seems like a bug in the firmware as Zhop height doesn't get cancelled after the print finishes without unretracting.

Armadillo1, 2019-02-17:
P.S. That's using 2.02 firmware.

T3P3, 2020-10-22:
Believed fixed in 2.03 - please reopen the issue if not.

closed 2020-10-22 by T3P3