Duet3D/RepRapFirmware #268: Request:  Splash Screen on Duet Maestro 12864 Display

shoreliner, 2019-02-27:
While I know it's possible to add a splash screen to a paneldue by attaching a bitmap to the tail of the PD firmware itself, this doesn't seem to be the case with The Duet Maestro and 12864 display.

Besides the ability to call a binary bitmap and have it displayed on startup, this command could be useful for other types of graphical popups on the LCD.

This would be a great feature and add an extra element to the user experience.

Thanks,

David

dc42, 2019-02-27:
That could be implemented as an optional menu file called 'splash', which if present is loaded. it cold contain both graphics and text. It would be displayed for e.g. 5 seconds before 'main' is loaded.

shoreliner, 2019-02-28:
After our discussion about a smaller font I tried going through the motions but found getting much smaller to be incredibly difficult.

That said, it seems like a better idea to conveying advanced concepts with very little real estate could be popups of diagrams, etc. The quad loading sequence is something, for example, that we'd utilize this for.  A visual aid versus all-text instructions.

Cheers,

David

shoreliner, 2019-02-28:
Not to creep too far away, but one more thing to add:  Say during this splash command(directly after power on) we could give the user the option to hold down the rotary encoder's button to access a recovery/maintenance menu.  This would primarily be something used in house but I Could see others utilizing it as well.

T3P3, 2025-05-07:
not doing on Maestro

closed 2025-05-07 by T3P3