Duet3D/RepRapFirmware #271: [2.03beta2] Stall-detection is not working properly

ikarisan, 2019-03-03:
I am using the Duet2Wifi on my Anet AM8 printer and the stall detection is not working properly since I upgraded to 2.03beta2. I was able to determine the following problem. Sometimes the X/Y axis initially is moving to the wrong direction.

In other words, the firmware detects the triggering of an endstop even though the axen are in the middle. They are just moving 5mm and then they think they are homed.

Has anything been changed in the sensitivity of the stall-detection?

Firmware: 2.03beta2 (2019-02-16b1)

The endstops are configured like this:

M574 X1 Y1 S3
M915 X Y S5 F0 R0

ctilley83, 2019-03-04:
according to the release notes:

`Caution: some of the code that supports stall-detection endstops has been rewritten. If your printer uses stall detection endstops, test homing carefully.`

You're going to have to optimize currents, speeds and sensitivity related to homing using stall detection. Try increasing the S parameter of your M915 command.

ikarisan, 2019-03-08:
After increasing S from 5 to 6 the initial direction sometimes is still wrong. And with S6 the stall detection does not trigger and the belt skips. :(

ctilley83, 2019-03-09:
@ikarisan Reread the docs on stall detection and senseless homing.  There are also other variables involved such as an optimal feedrate and current reduction. Try lowering current when you increase S.

dc42, 2019-11-08:
Please use the forum at duet3d.com for support.

closed 2019-11-08 by dc42