Duet3D/RepRapFirmware #281: [FR] Adding support for the Pyr0-Piezo Z-Probe (adding uart/i2c comms)

pyr0ball, 2019-04-14:
### Description

I've been working on a self-calibrating piezo sensor for a while, and I'm nearing completion, so I wanted to reach out and start working on code integration with Marlin. 

Information and sources can be found here: https://github.com/pyr0ball/pyr0piezo

### Requirements from Marlin

The sensor has been designed with UART and I2C input to allow the user to adjust the auto-calibration parameters on-the-fly. These sensors imitate a standard active-low endstop signal for it's output, so they can work out-of-the-box without these inputs, but there are a number of possible benefits for being able to adjust the sensitivity of the circuit:

- Different sized piezo disks produce different amplitude signals
- Decrease sensitivity during X/Y movements while probing, increase once probing Z
- Different mechanical structures for the piezo mount effect the signal amplitude

This integration could be implemented using GCode commands, and/or an LCD menu option

#### Input Syntax

To set the auto-tuning parameters using serial input, use the following:

* To change trigger active duration: TRG_D [integer for milliseconds]
* To change gain factor: GAIN_F [integer for gain state - see note*]
* To change sensor input pullup vRef high threshold: VADJH [float value]
* To change sensor input pullup vRef low threshold: VADJL [float value]
* To change comparator trigger high threshold: VCOMPH [float value]
* To change comparator trigger high threshold: VCOMPL [float value]

These commands should be wrapped in this format: 
`<CMD, INT, FLOAT>`

You must include the unused variable for each instance. 

Examples:
`<GAIN_F, 3, 0.00>`
`<VADJH, 0, 2.35>`

*Note for Gain Factor:

The gain STATE is representative of these values:
* 0 = 3x
* 1 = 3.5x
* 2 = 4.33x
* 3 = 6x
* 4 = 11x

T3P3, 2020-10-20:
Have you tried using the I2C commands:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M260_i2c_Send_and_or_request_Data

pyr0ball, 2020-10-20:
@T3P3 While that's doable, it's not a clean integration. I'd like to see something like a configuration menu similar to the options available to the TMC drivers, as this sensor has a lot of knobs and buttons that can be tweaked over the programming interface.

Here's the latest info on that configuration interface: https://docs.pyroballpcbs.com/tutorials/config/pyr0piezo-i2c/pyr0piezo-i2c/\

T3P3, 2020-10-20:
Thanks for the update. I think as a first pass using M260 as a macro should be tested. If it works well then I can see a visual plugin to DWC to allow for the sensor to be set. Are the setting persistent? if not then calling the macro from config.g to setup the sensor on boot would be a good idea.

Either way best to discuss this on the forum first to reach consensus on the way forward before submitting a PR.

pyr0ball, 2020-10-20:
The sensor includes an EEPROM, so yes they're persistent.

@T3P3 Should I nekro [the original thread about this](https://forum.duet3d.com/topic/16304/i2c-communication-with-either-the-duet3-or-hopefully/9?_=1603217756956) or start a new one?

T3P3, 2020-10-20:
I will replay to that one

pyr0ball, 2023-05-04:
I'm still getting orders on my sensors, so I think there's still some requirement

T3P3, 2024-03-06:
Not Planned