Duet3D/RepRapFirmware #284: MQTT Status Update / MQTT power control

Nightreaver, 2019-05-10:
Hello

I have seen MQTT files in the source code, but couldn't find any reference what to do with it yet.

I'd really like to add power control to Duet2Wifi but not via ext-PIN but via MQTT.

So I wonder (if not already available) if its possible to implement either of 2 options:

1. and MQTT setup + 2 commands, to turn on/off the printer before and after the print. It can be any command, and the settings are inside the firmware files (ie like wifi settings?). I assume they should be MQTT publish command? And maybe a separate MQTT query to figure out the status of the power supply via "ON" "OFF" status message of that query?

or

2. creating own MQTT topic, with the status of the print, in might include more information, like current layer height, duration etc, and a "POWER_DEMAND" status which will tell the power supply to be turned on / off. So before a print, the printer will publish a POWER_DEMAND=ON, and with some seperate setup (usually MQTT doenst make sense without anyways) i can create rules to turn the power on when POWER_DEMAND=ON, and off otherwise

I cannot guess the work load you might have with either of these options. I might say that everyone might have a bigger benefit from an own MQTT topic which includes lost of status information. But I also assume that its much easier for others work with a simple MQTT turn on/off that is just sent by the duet.

Either one is fine to me :)

Cheers

Tasmota has a good MQTT publishing system, it may be "reused", if there a no license issues of course. Thats kind of what I am thinking of for option #2 - https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT

T3P3, 2020-10-22:
This would be a useful addition to DSF using a plugin. supporting one of the (very) many messaging methods directly from RRF is unlikely

closed 2020-10-22 by T3P3

Nightreaver, 2020-10-22:
Hey, can you point to a "How to create a plugin" Guide or official documentation about plugins?

T3P3, 2020-10-22:
Not yeat, @chrishamm is working on some documentation. either way it will be for  [Duet Software Framework](https://github.com/Duet3D/DuetSoftwareFramework) not RepRapFirmware.

Nightreaver, 2020-10-22:
Okay, thank you!