Duet3D/RepRapFirmware #285: 2.03RC2 bed level issue!!!

TheMrCaveman, 2019-05-14:
With 2.03 bed level is alway different. Everytime i do a bed level it gives me a different height. See pictures. This is 2 times leveling after each other. 

This works flawles in 2.02, so I am wondering what is the cause of this issue.

2.03 is because of this isue a pain to have a good first layer.

[image]
[image]

T3P3, 2019-05-14:
It's best to ask for support on forum.duet3d.com

SonicDaHouseCat, 2019-05-25:
The same problem here with Duet 0.6 where a 9-factor calibration is performed before printing. Tried 2.03RC2 and 2.03RC1+1. Will downgrade and wait for 2.03RC3.

dc42, 2019-05-25:
Did you use babystepping? I have recently fixed a bug in 2.03RC2 (probably present in some earlier releases too) whereby the current babystepping offset was applied in the wrong direction after certain operations, such as bed probing.

SonicDaHouseCat, 2019-05-25:
Yeah, I used babystepping. I will test it in the next RC version.

closed 2020-10-22 by T3P3