Duet3D/RepRapFirmware #289: missing function

cdedwards, 2019-05-21:
Yesterday I attempted to setup RRF to compile. I downloaded the recommended programs. Downloaded the dev branches of each of the projects and attempted to compile. recieved the following error about GetStepInterval() missing in DDARing. Not sure what I'm missing at this point.

arm-none-eabi-g++ -D__SAM3X8E__ -DDUET_06_085 -D_XOPEN_SOURCE -I"C:\eclipse\Firmware\CoreNG\cores\arduino" -I"C:\eclipse\Firmware\CoreNG\libraries\Flash" -I"C:\eclipse\Firmware\CoreNG\libraries\SharedSpi" -I"C:\eclipse\Firmware\CoreNG\libraries\Storage" -I"C:\eclipse\Firmware\CoreNG\libraries\Wire" -I"C:\eclipse\Firmware\CoreNG\asf" -I"C:\eclipse\Firmware\CoreNG\asf\common\utils" -I"C:\eclipse\Firmware\CoreNG\asf\common\services\clock" -I"C:\eclipse\Firmware\CoreNG\asf\sam\drivers" -I"C:\eclipse\Firmware\CoreNG\asf\sam\services\flash_efc" -I"C:\eclipse\Firmware\CoreNG\asf\sam\utils" -I"C:\eclipse\Firmware\CoreNG\asf\sam\utils\cmsis\sam3x\include" -I"C:\eclipse\Firmware\CoreNG\asf\sam\utils\header_files" -I"C:\eclipse\Firmware\CoreNG\asf\sam\utils\preprocessor" -I"C:\eclipse\Firmware\CoreNG\asf\thirdparty\CMSIS\Include" -I"C:\eclipse\Firmware\CoreNG\variants\duet" -I"C:\eclipse\Firmware\RepRapFirmware\src" -I"C:\eclipse\Firmware\RepRapFirmware\src\Duet" -I"C:\eclipse\Firmware\RepRapFirmware\src\Duet\Lwip" -I"C:\eclipse\Firmware\RepRapFirmware\src\Duet\EMAC" -I"C:\eclipse\Firmware\RRFLibraries\src" -O2 -Wall -c -std=gnu++17 -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -nostdlib -Wdouble-promotion -fsingle-precision-constant "-Wa,-ahl=StepTimer.s" -MMD -MP -MF"src/Movement/StepTimer.d" -MT"src/Movement/StepTimer.o" -o "src/Movement/StepTimer.o" "../src/Movement/StepTimer.cpp"
In file included from ../src/Movement/Move.h:13:0,
                 from ../src/Movement/DriveMovement.cpp:10:
../src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
../src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
In file included from ../src/Movement/Move.h:13:0,
                 from ../src/Movement/StepTimer.cpp:11:
../src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
../src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
In file included from C:\eclipse\Firmware\RepRapFirmware\src/Movement/Move.h:13:0,
                 from ../src/Movement/Kinematics/HangprinterKinematics.cpp:12:
C:\eclipse\Firmware\RepRapFirmware\src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
C:\eclipse\Firmware\RepRapFirmware\src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
In file included from ../src/Movement/Move.h:13:0,
                 from ../src/Movement/DDA.cpp:11:
../src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
../src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
In file included from C:\eclipse\Firmware\RepRapFirmware\src/Movement/Move.h:13:0,
                 from ../src/Movement/Kinematics/LinearDeltaKinematics.cpp:10:
C:\eclipse\Firmware\RepRapFirmware\src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
C:\eclipse\Firmware\RepRapFirmware\src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
In file included from C:\eclipse\Firmware\RepRapFirmware\src/Movement/Move.h:13:0,
                 from ../src/Movement/Kinematics/ZLeadscrewKinematics.cpp:12:
C:\eclipse\Firmware\RepRapFirmware\src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
C:\eclipse\Firmware\RepRapFirmware\src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
In file included from ../src/Movement/DDARing.cpp:8:0:
../src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
../src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
src/Movement/subdir.mk:30: recipe for target 'src/Movement/DriveMovement.o' failed
make: *** [src/Movement/DriveMovement.o] Error 1
make: *** Waiting for unfinished jobs....
src/Movement/subdir.mk:30: recipe for target 'src/Movement/StepTimer.o' failed
make: *** [src/Movement/StepTimer.o] Error 1
src/Movement/Kinematics/subdir.mk:39: recipe for target 'src/Movement/Kinematics/HangprinterKinematics.o' failed
make: *** [src/Movement/Kinematics/HangprinterKinematics.o] Error 1
src/Movement/Kinematics/subdir.mk:39: recipe for target 'src/Movement/Kinematics/ZLeadscrewKinematics.o' failed
make: *** [src/Movement/Kinematics/ZLeadscrewKinematics.o] Error 1
src/Movement/Kinematics/subdir.mk:39: recipe for target 'src/Movement/Kinematics/LinearDeltaKinematics.o' failed
make: *** [src/Movement/Kinematics/LinearDeltaKinematics.o] Error 1
src/Movement/subdir.mk:30: recipe for target 'src/Movement/DDARing.o' failed
make: *** [src/Movement/DDARing.o] Error 1
src/Movement/subdir.mk:30: recipe for target 'src/Movement/DDA.o' failed
make: *** [src/Movement/DDA.o] Error 1
In file included from ../src/Movement/Move.h:13:0,
                 from ../src/Movement/Move.cpp:36:
../src/Movement/DDARing.h: In member function 'uint32_t DDARing::GetStepInterval(size_t, uint32_t) const':
../src/Movement/DDARing.h:130:35: error: 'const class DDA' has no member named 'GetStepInterval'
  return (cdda != nullptr) ? cdda->GetStepInterval(axis, microstepShift) : 0;
                                   ^~~~~~~~~~~~~~~
src/Movement/subdir.mk:30: recipe for target 'src/Movement/Move.o' failed
make: *** [src/Movement/Move.o] Error 1

15:05:09 Build Failed. 25 errors, 0 warnings. (took 7s.886ms)

dc42, 2019-05-23:
It compiles for me. In the latest source, the call to that function is in conditional #if HAS_SMART_DRIVERS so it should not be active in the Duet085 build.

cdedwards, 2019-05-26:
I've gotten further, but still not compiled completely.  I think part of the issue is that I'm attempting to compile under windows.. I found where to change the compile target in eclipse but found that I now had other errors. I reinstalled all the sources, and eclipse would not load the .cproject files leaving me only with the default. And that would not work at all. Eventually I got it working again by using git to download the sources. At the linking stage I ran into a stumbling block. I tried switching from the eclipse-workspace directory to the git directory and ran into the same issue with .cproject not loading. syscalls.o is not in the eclipse-workspace directory but instead is in the CoreNG directory. I have gotten it to complete by creating the directory it was expecting and copying the file into that location in eclipse-workspace.

17:58:41 **** Incremental Build of configuration Duet2_RTOS for project RepRapFirmware ****
make -j8 all 
Building target: Duet2CombinedFirmware.elf
Invoking: Cross G++ Linker
arm-none-eabi-gcc -L"C:\Users\Colten Edwards\git\CoreNG\SAM4E8E" -L"C:\Users\Colten Edwards\git\FreeRTOS\SAM4E" -L"C:\Users\Colten Edwards\git\RRFLibraries\SAM4E_RTOS" -Os --specs=nano.specs -Wl,--gc-sections -Wl,--fatal-warnings -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -T"C:\Users\Colten Edwards\git\CoreNG\variants\duetNG\linker_scripts\gcc\flash.ld" -Wl,-Map,"C:\Users\Colten Edwards\git\RepRapFirmware\Duet2_RTOS/Duet2CombinedFirmware.map" -o "Duet2CombinedFirmware.elf" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "C:\Users\Colten Edwards\eclipse-workspace/CoreNG/SAM4E8E/cores/arduino/syscalls.o"  ./src/Tools/Filament.o ./src/Tools/Tool.o  ./src/Storage/CRC32.o ./src/Storage/FileInfoParser.o ./src/Storage/FileStore.o ./src/Storage/MassStorage.o  ./src/ObjectModel/ObjectModel.o  ./src/Networking/W5500Ethernet/Wiznet/Internet/DHCP/dhcp.o  ./src/Networking/W5500Ethernet/Wiznet/Ethernet/W5500/w5500.o  ./src/Networking/W5500Ethernet/Wiznet/Ethernet/WizSpi.o ./src/Networking/W5500Ethernet/Wiznet/Ethernet/socketlib.o ./src/Networking/W5500Ethernet/Wiznet/Ethernet/wizchip_conf.o  ./src/Networking/W5500Ethernet/W5500Interface.o ./src/Networking/W5500Ethernet/W5500Socket.o  ./src/Networking/ESP8266WiFi/WiFiInterface.o ./src/Networking/ESP8266WiFi/WiFiSocket.o ./src/Networking/ESP8266WiFi/WifiFirmwareUploader.o  ./src/Networking/FirmwareUpdater.o ./src/Networking/FtpResponder.o ./src/Networking/HttpResponder.o ./src/Networking/Network.o ./src/Networking/NetworkBuffer.o ./src/Networking/NetworkResponder.o ./src/Networking/TelnetResponder.o ./src/Networking/UploadingNetworkResponder.o  ./src/Movement/StepperDrivers/DriverMode.o ./src/Movement/StepperDrivers/TMC22xx.o ./src/Movement/StepperDrivers/TMC2660.o ./src/Movement/StepperDrivers/TMC51xx.o  ./src/Movement/Kinematics/CoreKinematics.o ./src/Movement/Kinematics/HangprinterKinematics.o ./src/Movement/Kinematics/Kinematics.o ./src/Movement/Kinematics/LinearDeltaKinematics.o ./src/Movement/Kinematics/PolarKinematics.o ./src/Movement/Kinematics/RotaryDeltaKinematics.o ./src/Movement/Kinematics/ScaraKinematics.o ./src/Movement/Kinematics/ZLeadscrewKinematics.o  ./src/Movement/BedProbing/Grid.o ./src/Movement/BedProbing/RandomProbePointSet.o  ./src/Movement/DDA.o ./src/Movement/DDARing.o ./src/Movement/DriveMovement.o ./src/Movement/Move.o ./src/Movement/StepTimer.o  ./src/Libraries/sha1/sha1.o  ./src/Libraries/Fatfs/diskio.o ./src/Libraries/Fatfs/fattime_rtc.o ./src/Libraries/Fatfs/ff.o ./src/Libraries/Fatfs/ffunicode.o  ./src/Heating/Sensors/CpuTemperatureSensor.o ./src/Heating/Sensors/CurrentLoopTemperatureSensor.o ./src/Heating/Sensors/DhtSensor.o ./src/Heating/Sensors/RtdSensor31865.o ./src/Heating/Sensors/SpiTemperatureSensor.o ./src/Heating/Sensors/TemperatureSensor.o ./src/Heating/Sensors/Thermistor.o ./src/Heating/Sensors/ThermocoupleSensor31855.o ./src/Heating/Sensors/ThermocoupleSensor31856.o ./src/Heating/Sensors/TmcDriverTemperatureSensor.o  ./src/Heating/FOPDT.o ./src/Heating/Heat.o ./src/Heating/HeaterProtection.o ./src/Heating/Pid.o ./src/Heating/TemperatureError.o  ./src/Hardware/DmacManager.o ./src/Hardware/I2C.o ./src/Hardware/IoPorts.o  ./src/GCodes/GCodeBuffer.o ./src/GCodes/GCodeInput.o ./src/GCodes/GCodeMachineState.o ./src/GCodes/GCodeQueue.o ./src/GCodes/GCodes.o ./src/GCodes/GCodes2.o ./src/GCodes/GCodes3.o ./src/GCodes/RestorePoint.o  ./src/FilamentMonitors/Duet3DFilamentMonitor.o ./src/FilamentMonitors/FilamentMonitor.o ./src/FilamentMonitors/LaserFilamentMonitor.o ./src/FilamentMonitors/PulsedFilamentMonitor.o ./src/FilamentMonitors/RotatingMagnetFilamentMonitor.o ./src/FilamentMonitors/SimpleFilamentMonitor.o  ./src/Fans/DotStarLed.o ./src/Fans/Fan.o ./src/Fans/Tacho.o  ./src/DuetNG/DueXn.o ./src/DuetNG/SX1509.o  ./src/Display/ST7920/glcd11x14.o ./src/Display/ST7920/glcd7x11.o ./src/Display/ST7920/lcd7920.o  ./src/Display/Display.o ./src/Display/Menu.o ./src/Display/MenuItem.o ./src/Display/RotaryEncoder.o  ./src/CAN/CanInterface.o ./src/CAN/CanMessageBuffer.o ./src/CAN/CanSender.o  ./src/Logger.o ./src/OutputMemory.o ./src/Platform.o ./src/PortControl.o ./src/PrintMonitor.o ./src/RepRap.o ./src/RepRapFirmware.o ./src/Roland.o ./src/Scanner.o ./src/SoftTimer.o ./src/Spindle.o ./src/Tasks.o ./src/ZProbe.o ./src/ZProbeProgrammer.o   -lCoreNG -lFreeRTOS -lRRFLibraries -Wl,--end-group -lm
arm-none-eabi-gcc: error: C:\Users\Colten Edwards\eclipse-workspace/CoreNG/SAM4E8E/cores/arduino/syscalls.o: No such file or directory
makefile:70: recipe for target 'Duet2CombinedFirmware.elf' failed
make: *** [Duet2CombinedFirmware.elf] Error 1

17:58:44 Build Failed. 2 errors, 0 warnings. (took 3s.703ms)

dc42, 2019-11-08:
Looks like you haven't built the SAM4E config of CoreNG. If you require further support, please use the forum at duet3d.com.

closed 2019-11-08 by dc42