Duet3D/RepRapFirmware #290: Feature request: Show mesh Z offset modifier next to Z.

jphillips, 2019-05-26:
Having to load and unload my heightmap to see what the current offset is at the current position is very slow. I do it to verify my probe and tool offsets lineup perfectly with the heightmap. I put a coin cell battery on the plate, run an mesh grid comp. Then use this slow load unload method with my tool selected to find the spot with the highest point. 

In the end, it didn't line up for me, possibly another bug. 
Try it out. (do a bltouch with its own offset from the HEAD, and a tool with its own offset from the HEAD) 
When the tool is selected, if you go to the area you recorded the highest point in the bed, you will find that the more you minus out the tool offset by stepping the x/y, the higher the point gets until the tool has no offset. But maybe I'm doing it wrong.

T3P3, 2023-05-04:
Please discuss in a forum thread if the requirement still exists.

dc42, 2023-05-04:
I suggest we add the current Z compensation amount to the object model if it is easy to do.

T3P3, 2026-09-16:
Forum threads:
https://forum.duet3d.com/topic/36053
https://forum.duet3d.com/topic/31181