Duet3D/RepRapFirmware #299: Add support for ZeroConf/Bonjour network discovery

TheBum, 2019-07-13:
It would be really handy to have this feature for Mac and Linux users so that we don’t have to deal with setting up a local DNS server or use hard-coded IP addresses.

Renha, 2019-07-23:
isn't that already supported? I mean, my Ashwood machine is discoverable as ashwood.local.

TheBum, 2019-07-23:
Maybe Ashwood is using custom firmware. I just know that I can’t get to my Crane using cranequad.local (the defined machine name). The web page for the OctoPi connected to my Snapmaker comes right up when I go to octopi.local.

TheBum, 2019-07-23:
I just looked through the source code and it appears mDNS was disabled due to a buggy implementation. I have not discovered when it was disabled, but I suspect that your firmware version predates the change.

Maybe it's time to revisit it with a newer version of the support code?

alexjx, 2019-07-23:
I could connect with the domain name without problem... what do you mean by "buggy implementation"?

chrishamm, 2019-07-24:
It is already supported on the Duet WiFi because it could be easily enabled there, but an implementation for the Duet Ethernet/Maestro is not trivial and will require extra refactoring. It is planned, but there is no timeline for the implementation yet.

chrishamm, 2019-10-17:
It's implemented in the latest v2 and v3 releases.

closed 2019-10-17 by chrishamm