Duet3D/RepRapFirmware #300: Warning: the height map was loaded when the current Z=0 datum was not determined probing.

TheNetStriker, 2019-07-18:
In the firmware version 2.03 I get the following warning every time I tried to home the z axis:

Warning: the height map was loaded when the current Z=0 datum was not determined probing. This may result in a height offset

I already executed the mesh bed leveling macro multiple times, but the warning still appears.

I was able to trace the warning back to the command G29 S1 in the homez.g file. In the Discord forum I got the hint to remove the G29 S1 command from config.g file, but that also didn't help.

What could be the cause for this warning?

violetbp, 2020-11-10:
For me this I only get the error when I try to load the mesh settings before doing a home which is why they recommended removing the G29 S1, from what I know that's the only thing that would cause that

T3P3, 2023-05-04:
Please discuss on the forum if this issues still exists in RRF 3.4.5

closed 2023-05-04 by T3P3