Duet3D/RepRapFirmware #302: 2.04RC1 auto levelling results in overall z offset

hihearme, 2019-07-20:
Hi, when I was trying the new 2.04RC1 firmware, I realized that whenever I issues a G29 command, the height map has a positive z offset ~0.2 mm through out the probed area. Issuing multiple G29 commands in a row results in 0.2 - 0.3 mm positive z offset each time, producing higher and higher height maps. Switching back to firmware 2.03 appears to resolve the issue.

cdedwards, 2019-07-20:
sounds like the previous map is still active while you are probing.

hihearme, 2019-07-20:
> sounds like the previous map is still active while you are probing.

The z offset happens upon the first G29 cycle even after I deleted the old height map file and rebooted the system. Plus, I tried adding G29 S2 at the beginning of my homeall and homez files without any help.

T3P3, 2020-10-22:
Please check with the latest firmware and see if this is still an issue

closed 2020-10-22 by T3P3