Duet3D/RepRapFirmware #303: 2.04RC1 G29 missing points

cdedwards, 2019-07-21:
my printer's BLTouch is 65mm from the nozzle in X. My bed is 335mm on the X and I'm able to reach the full bed with my probe if I set M208 to a larger value than 335mm. G32 doesn't have a problem with reaching the edge in order to probe, but G29 fails.

7/20/2019, 9:49:19 PM | Warning: Skipping grid point (280.0, 40.0) because Z probe cannot reach it 
Warning: Skipping grid point (296.0, 40.0) because Z probe cannot reach it 
Warning: Skipping grid point (312.0, 40.0) because Z probe cannot reach it 
Warning: Skipping grid point (328.0, 40.0) because Z probe cannot reach it 
Warning: Skipping grid point (328.0, 56.0) because Z probe cannot reach it 
Warning: Skipping grid point (312.0, 56.0) because Z probe cannot reach it 
Warning: Skipping grid point (296.0, 56.0) because Z probe cannot reach it 
Warning: Skipping grid point (280.0, 56.0) because Z probe cannot reach it

bisciano, 2020-03-14:
I also have the same problem

Delta radius 165
M557 R160 S20

Warning: Skipping grid point (120.0, -100.0) because Z probe cannot reach it
--
15.03.2020, 01:23:25 | Warning: Skipping grid point (-100.0, -120.0) because Z probe cannot reach it Warning: Skipping grid point (-120.0, -100.0) because Z probe cannot reach it
15.03.2020, 01:22:54 | Warning: Skipping grid point (60.0, -140.0) because Z probe cannot reach it Warning: Skipping grid point (100.0, -120.0) because Z probe cannot reach it
15.03.2020, 01:22:36 | G29Warning: Skipping grid point (-60.0, -140.0) because Z probe cannot reach it

T3P3, 2020-10-22:
You are asking the printer to probe a point outside of the bed area, given the z probe offset.

Please use the forum for support with configuration issues (forum.duet3d.com)

closed 2020-10-22 by T3P3