Duet3D/RepRapFirmware #307: not all axes are homing

ulcast, 2019-08-21:
Hi, when i try to home all axes, if they are misaligned, there is always one axe that doesn't home and i get the error "insufficient axes homed". i mean that when the first two axes are homed the third stops moving and doesnt't home.
if i press "home all axes" also the last axe homes.

if the axes are aligned (the nozzle is in the middle) it homes with no problem.

any solution?

thanks!

this is my homedelta.g

; homedelta.g
; called to home all towers on a delta printer
;
; generated by RepRapFirmware Configuration Tool v2 on Tue Aug 20 2019 23:53:32 GMT+0200 (Ora legale dell’Europa centrale)
G91                        ; relative positioning
G28
G1 S1 X246 Y246 Z246 F1800 ; move all towers to the high end stopping at the endstops (first pass)
G1 X-5 Y-5 Z-5 F1800 S2    ; go down a few mm
G1 S1 X10 Y10 Z10 F360     ; move all towers up once more (second pass)
G1 Z-5 F6000               ; move down a few mm so that the nozzle can be centred
G90                        ; absolute positioning
G1 X0 Y0 F6000             ; move X+Y to the centre

StephenRC, 2019-08-22:
You need to have the initial axis movement to be longer than the
maxium the axis can be physically moved.
 
On Wed, 21 Aug 2019 10:41:07 -0700, ulcast <notifications@github.com>
had a flock of green cheek conures squawk out:

> Hi, when i try to home all axes, if they are misaligned, there is always one axe that doesn't home and i get the error "insufficient axes homed". i mean that when the first two axes are homed the [...]

ulcast, 2019-08-22:
How can I do that?

Zaibot, 2019-08-22:
I had this problem to for cartesian. From the top of my head there were two locations I had to change. For XY in config.g and Z in homez.g or something.

See the FAQ:
https://duet3d.dozuki.com/Wiki/FAQ#Section_When_I_try_to_home_my_printer_sometimes_the_carriage_or_carriages_move_only_part_of_the_way_towards_the_homing_switches_and_I_have_to_command_it_to_home_again_to_make_it_go_all_the_way

> If your printer is a delta, open the homedelta.g file in the System Editor of the web interface. Look for the first G1 S1 line in that file.

StephenRC, 2019-08-22:
Start by doubling the 246 you have on the G1 for the affected axes.

For example:
G1 S1 X500 Y500 Z500 F1800 ; move all towers to the high end stopping
at the endstops (first pass)

On Thu, 22 Aug 2019 05:20:18 -0700, ulcast <notifications@github.com>
had a flock of green cheek conures squawk out:

>How can I do that?

ulcast, 2019-08-24:
Thanks! solved!

closed 2020-10-22 by T3P3