Duet3D/RepRapFirmware #310: Tool select (Tx) ignores pre/post scripts

orcinus, 2019-08-26:
After homing all axes from the web UI, tool changes from a job started via PanelDue simply ignore all the pre/post scripts. Tools get selected, but the scripts don't get run.

If i home axes from PanelDue, however, jobs run from the GUI or the PanelDue will execute the tool change scripts just fine.

I have "Run free macro", "Run pre macro" and "Run post macro" turned on in the Web UI.

I'm not sure if this is a RRF bug, a PanelDue bug, or a web interface bug (or a combination of all three).

orcinus, 2019-08-28:
It just occurred to me that maybe this happens when you hit print or select a tool too quickly after a fresh boot or a reboot. Does it take a while for RRF to load or cache the system gcode macros?

chrishamm, 2019-08-28:
Tool change scripts are only processed after config.g has been fully read and if all axes are homed. Are you trying to change the tool in a homing macro?

orcinus, 2019-08-29:
No, in a print job gcode file loaded and run immediately after homing.

orcinus, 2019-08-30:
I'm not exactly sure what changed, but couldn't reproduce it anymore in the past 3 days.
I can close and reopen if it happens again?

chrishamm, 2019-10-17:
I'll close it. If it happens again, either reopen this issue or create a new one.

closed 2019-10-17 by chrishamm