Duet3D/RepRapFirmware #323: WebControl 2.02 Extrusion Control dosn't work

stubbeDK, 2019-10-10:
[image]

The buttons Extrude or Retract isn't enabled after heating up the hotend why? 
Have always worked. Can still do it from the console :)

chrishamm, 2019-10-17:
I could reproduce this problem and I am working on a fix. In the future, please report DWC-related problems on the official DWC repo here: https://github.com/chrishamm/DuetWebControl/issues

stubbeDK, 2019-10-17:
> 
> 
> I could reproduce this problem and I am working on a fix. In the future, please report DWC-related problems on the official DWC repo here: https://github.com/chrishamm/DuetWebControl/issues

Sorry about the miss placement, i know for next time and thanks :)

closed 2019-11-08 by dc42