Duet3D/RepRapFirmware #325: Feature Request: Save Extruder Temperature

KyleMaas, 2019-10-11:
I just added a rotating magnet filament sensor to my machine, since I have certain filaments that I print with which tend to hang up due to bulges in the filament.  The machine pauses the print as it should, and all works quite well.

Except when I'm not paying attention and it pauses for an extended period of time.

It would be very helpful if there was a way to have the machine shut down the extruder when it pauses, then bring it back up to temperature before resuming.  Only the extruder - I would like to leave the bed on so parts don't loosen from the bed during the pause.  But if it sits there paused for an extended period of time, I'd like to have the nozzle turned off.

I have not found a way to do that in supported G-Code in the pause.g and resume.g files, but maybe I'm missing something.  Regardless of mechanism, whether it's a pair of G-Code commands which could be added, a variation of the push/pop commands, or something configurable using config.g, my goal is to shut off the hot end when the machine is paused.

Is this possible?

T3P3, 2023-05-04:
This is now possible by using the Event system in RRF 3.4 and later

closed 2023-05-04 by T3P3

KyleMaas, 2023-06-15:
Is it possible you could give an example of how to do that?