Duet3D/RepRapFirmware #326: Monitoring V_IN in graph Window

pipersw, 2019-10-15:
Allow to see voltage drop while printing

pipersw, 2020-01-06:
Reopen because DWC need to have a new variable in RRF in order to display the input voltage V_IN.
So, it is possible to add a sensor_type for V_IN in M308 GCODE ?

pipersw, 2020-01-14:
https://github.com/chrishamm/DuetWebControl/issues/168

pipersw, 2020-11-23:
Hi !
it could be implemented in the next beta software ?

dc42, 2020-11-23:
It would also be necessary for DWC to understand that it is displaying units other than temperature, so that it can display an additional Y axis. If a 24V VIN was displayed on the same scale as the temperatures, you wouldn't see the fluctuations.

pipersw, 2020-12-02:
> It would also be necessary for DWC to understand that it is displaying units other than temperature, so that it can display an additional Y axis. If a 24V VIN was displayed on the same scale as the temperatures, you wouldn't see the fluctuations.

The best will be a second graph for VIN

chrishamm, 2020-12-02:
Extra units are already supported in DWC, they may be specified using squared brackets in the sensor name, e.g. "DHT Humidity [%]" or in a future case "Input voltage [V]". However secondary Y axes in the temperature chart are not supported yet.