Duet3D/RepRapFirmware #333: G38.2 to G38.5 positions are always in absolute machine coordinates

jannau, 2019-11-03:
The G38.x commands added in #324 interpret target position always absolute machine coordinates. Running 'RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 3.0beta12' on a cartesian CNC machine XYZ zero in upper left back corner. When trying following probe command sequence without triggering the probe:
```
M114
X:-10.000 Y:-10.000 Z:0.000 E:0.000 E0:0.0 E1:0.0 E2:0.0 Count -4000 -4000 0 Machine -10.000 -10.000 0.000 Bed comp 0.000
G91
G38.3 X-5 P1
M114
X:-5.000 Y:-10.000 Z:0.000 E:0.000 E0:0.0 E1:0.0 E2:0.0 Count -2000 -4000 0 Machine -5.000 -10.000 0.000 Bed comp 0.000
```
I would have expected the probe move to end at `X:-15`. When setting a workspace offset for WCS G54:
```
M114
X:0.000 Y:0.000 Z:0.000 E:0.000 E0:0.0 E1:0.0 E2:0.0 Count -4000 -4000 0 Machine -10.000 -10.000 0.000 Bed comp 0.000
G91
G38.3 X-5 P1
M114
X:5.000 Y:0.000 Z:0.000 E:0.000 E0:0.0 E1:0.0 E2:0.0 Count -2000 -4000 0 Machine -5.000 -10.000 0.000 Bed comp 0.000
```
This degrades the position arguments to a direction argument if G38.x is used in a macro since the axis' minimum or maximum are the only positions which guarantee a move in the right direction. This is also quite unsafe since it could result in a crash when the probe erroneously fails to trigger.

Is there a way to prevent further execution of a macro in the case a G38.2 or G38.4 fails? consider following macro:
```
G91
G38.2 Z-10 P1
G10 L20 P1 Z0
G90
G53 G1 Z0
```
It would be onvenient to update the Z work coordinate only if the probe triggered during the G38.2.

T3P3, 2023-05-04:
There have been a lot of changes in RRF since this issue was raised, Please open a discussion on the forum if this is still a requirement.

Aquilux, 2023-07-20:
I've run into this issue trying to make macros myself. Should I make a forum discussion?

dc42, 2025-07-31:
Re-opened and assign to milestone 3.6.1

WheelerCC, 2025-08-17:
My approach has been to use the following macro, defined once per face (front/left/back/right) to probe

```
; 0:/macros/XYZ-Probe/locate-front-edge.g

var probeOffset = 0.891         

G91                         ; relative mode
G38.2 K1 Y{move.axes[1].userPosition+5} F200
echo result
if result != 0
    abort "Didn't find an edge (fast)"
;G38.2 K1 X10 F100  ; probe negative X direction
G0 Y-1      ; retract a bit
G38.2 K1 Y{move.axes[1].userPosition+1.5} F50
if result != 0
    abort "Didn't find an edge (slow)"
G0 Y-1      ; retract a bit
G90                         ; back to absolute mode

G53 G0 Z{global.Z_MAX}
G91                         ; relative mode
G0 Y{1+var.probeOffset}      ; move to where probe thinks 0 is on relevant axis
G90                         ; back to absolute mode

```

(Note I don't have this macro calculating/storing the WCS offset required to make the current tool position 0 in the given axis, I prefer to do that afterwards. This just handles the probing, and leave the tool at where the probe thinks 0 is)

dc42, 2025-08-20:
Fixed in 3.6-dev ready for 3.6.1 release.

closed 2025-08-20 by dc42

WheelerCC, 2025-09-21:
I can confirm that 3.6.1 behaves as I'd expect it when using relative coordinates. 

Note that after the fix, the workaround in my macro example above needs updating (especially the right and rear equivalent, as the `move.axes[1].userPosition+5` direction and max distance will be wrong! For left and front, the direction is correct by coincidence but the max distance is still wrong) . 3.6.1 and onwards, the macro along the lines of

```
; 0:/macros/XYZ-Probe/locate-front-edge.g

var probeOffset = 0.891         

G91                         ; relative mode
G38.2 K1 Y5 F200
echo result
if result != 0
    abort "Didn't find an edge (fast)"
;G38.2 K1 X10 F100  ; probe negative X direction
G0 Y-1      ; retract a bit
G38.2 K1 Y1.5 F50
if result != 0
    abort "Didn't find an edge (slow)"
G0 Y-1      ; retract a bit
G90                         ; back to absolute mode

G53 G0 Z{global.Z_MAX}
G91                         ; relative mode
G0 Y{1+var.probeOffset}      ; move to where probe thinks 0 is on relevant axis
G90                         ; back to absolute mode
```