Duet3D/RepRapFirmware #336: Laser doesn't turn on during G1 movement.

JyriT, 2019-11-08:
I ran in same situation with following code with Release 2.04 (Duet 2) and 1.25 (Duet 06/085).
Tested also with 3.0beta12 (2019-11-02b1) using M452 C"exp.heater3" Q200.
Config is in attached file.config.g
[config.g.txt](https://github.com/dc42/RepRapFirmware/files/3824324/config.g.txt)

Seems that heater3 output is turned off, or it doesn't turn on correctly while moving with G1.
gcode:
;Project bugbug
; *** START begin ***
G90
G21
M84 S0
G92 X0 Y0 Z0
G00 X0.0000 Y0.0000 Z5.0000

;No. 2: Engraving 1
G00 X25.7569 Y48.9645 Z5.0000
G00 Z0.5000
G01 Z0.0000 F1200
M3 S255
G01 X19.7518 Y30.6457 F750
;bug hits here. previus move is lasered ok, next 2 moves not.
;if run line by line (pasting one by one to console) all works ok.
G01 X1.1829 Y30.6942
G01 X16.1260 Y19.5851
;but next move from here is working ok again.
G01 X10.1260 Y20.5851
M5
G00 Z5.0000
G00 X0.0000 Y0.0000
M05

JyriT, 2019-11-10:
Here is another sample. Error hits in also while using G1 with S - parameter. 
Attached Gcode can be used to trigger bug. Happens somewhere around line 30. 
[peukku.nc.txt](https://github.com/dc42/RepRapFirmware/files/3828652/peukku.nc.txt)

T3P3, 2020-10-22:
Please test the latest firmware and open a discussion on the forum if it is still not working.

closed 2020-10-22 by T3P3