Duet3D/RepRapFirmware #342: Allow full-speed probing for analogue piezo probes.

tekkydave, 2019-12-16:
Currently the z probing is cut to one-third speed if the probe is defined as analogue. This is not appropriate for analogue piezo probes.
Temporarily increasing the z axis max speed is not a workround as it also increases the probe's lift speed, pushing it beyond the capability of the z axis motors. This can cause stalling.

Suggested solutions:
1. A way of specifying the amount of slowdown for analog sensors in M558
2. A separate M558 probe type for piezo probes with no slowdown applied.

Thank you :)

ahmetipkin, 2020-07-21:
I'm sorry I cannot check and verify this now; but after issuing M558 P1 ...; can you not configure the dive height and speed etc. parameters with [G31](https://duet3d.dozuki.com/Wiki/Gcode#Section_G31_Set_or_Report_Current_Probe_status) ?

I thought you can always use G31, whatever the probe type...

tekkydave, 2021-03-11:
RRF3.3beta2 resolves this issue due to changes in M558 F parameter. Piezo probes now dive at full speed.

closed 2021-03-11 by tekkydave