Duet3D/RepRapFirmware #343: RRF3: Z position display changes when heightmaps load/unload.

gtjoseph, 2019-12-17:
Without heightmap it loaded: Z` "machinePosition" : 15.200,`
The heightmap shows "0.145" for that specific XY location. so...
With the heightmap loaded: Z` "machinePosition" : 15.055,`
This with all axes at a standstill.

What's more, with the heightmap loaded, when the XY axes are moving but the Z axis is not, the reported Z position changes depending on the heightmap correction for the XY location the carriage is traversing.

This was never the case with RRF2.

Nibbels, 2020-04-07:
My post does not refer to the absolute same topic but It might be the same cause.
I did not want to open a second issue while this is stalled and unanswered. My issue would be:

"**RRF3: Z position display changes when changing a tool or pausing on a Toolchanger. The added Z offset is tool dependend and height map dependend. Pausing and unpausing cumulates the offsets**."

I documented my path towards my current knowledge here:
https://forum.e3d-online.com/threads/layer-heights-different-between-tools.3637/

This evening I recorded the video. It was the first try to make the movie. Sorry for the noise and uncoordinated guidance. But It is already late and I didnt want to delay this topic by making the absolute perfect video.
https://www.youtube.com/watch?v=VqTbf3GnA2o

Config and all the steps(commits) towards this config is found here:
https://github.com/Nibbels/e3d-toolchanger-config/tree/master
Just before posting this I refreshed all the content to be equal to my printers files.

The test part is uploaded as well: "3HeartTestTeil - T0-285 T1-285 T2-175 - use dryrun.gcode"
It has been created using simplify 3D. I use hardware retraction and filament configurations as seen within the githubs configuration.

Billtopia, 2020-04-07:
For the tool change you should be able to undo the tool offsets in the tfreeN.g file

Nibbels, 2020-04-08:
The bug seems to be gone when I dont use bed leveling.

Today I removed all the filament.
Then I manually pressed the Z endstop to print into the air.
And started the same print again.

https://youtu.be/osxBDhneF1M

What you see within the Video: The bug is gone and the physical offsets seem to be fine.
Changes are:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

In case someone thinks what ;G1 R2 Z0 has to do within tpostN.g: I previously added this as another test and it did not change my problem.
I once found that here: https://forum.duet3d.com/topic/9347/tool-offset-ignored-z-axis

Nibbels, 2020-04-08:
To return to the buggy behaviour I added 
G29 S1 ; Use the Z offset Mesh Map on the SD card
into start.g again.
[image]
And I put a measurement device on the bed.
[image]

And thats the full video which shows the behaviour and all the detailed z problems:
- pausing
- toolchanging
- correction at G1 Zxxx / layer change

https://youtu.be/aEVe9OVEAUo

Nibbels, 2020-04-30:
My issue seems to be solved with RRF 3.01 **RC11**

See here: https://forum.e3d-online.com/threads/layer-heights-different-between-tools.3637/page-3#post-36633 and prior.

And see
https://github.com/dc42/RepRapFirmware/blob/v3-dev/WHATS_NEW_RRF3.md
Bug fixes:
"Pause and resume sometimes caused a small Z shift if bed compensation was in use and the tool had an X or Y offset"
That bugfix as well fixed the tool change offset I reported.

@gtjoseph did your problem vanish with this version too?

In case we both (@gtjoseph) had the same kind of issue we probably can close this issue now.

Nibbels, 2020-04-30:
https://github.com/dc42/RepRapFirmware/commit/5dda6c9d5aaa24ea94f9246db73e702580e5d0dc

closed 2020-10-22 by T3P3