Duet3D/RepRapFirmware #346: MKS Sbase v1.3 installation

aceranic, 2019-12-28:
Hi,
I am very interested in trying RR Firmware on MKSSbase v1.3 but I am haviung issues with instalation.

For Smoothie installation it is required to add firmware.bin and config files to sd card and restart the board.

i have done configuration from https://sdavi.github.io/LPCConfigurator/ but have issue with which files are required and how to proceed now.

thanks,

aceranic, 2020-01-03:
Nothing?

cdedwards, 2020-01-03:
this RR version is for the Duet2/3 hardware and will not work on the MKS Sbase boards. Your looking for Smoothieware

aceranic, 2020-01-04:
So wrong place for post, thanks!!Poslano s pametnog telefona Samsung Galaxy.
-------- Izvorna poruka --------Šalje: Colten Edwards <notifications@github.com> Datum: 03. 01. 2020.  23:19  (GMT+01:00) To: dc42/RepRapFirmware <RepRapFirmware@noreply.github.com> Cc: aceranic <aceranic1@gmail.com>, Author <author@noreply.github.com> Predmet: Re: [dc42/RepRapFirmware] MKS Sbase v1.3 installation (#346) this RR version is for the Duet2/3 hardware and will not work on the MKS Sbase boards. Your looking for Smoothieware

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/dc42/RepRapFirmware/issues/346?email_source=notifications\u0026email_token=AEBDTJISOEYXV6PJMRY7SFTQ362WBA5CNFSM4KAKUD3KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEICG2KI#issuecomment-570715433",
"url": "https://github.com/dc42/RepRapFirmware/issues/346?email_source=notifications\u0026email_token=AEBDTJISOEYXV6PJMRY7SFTQ362WBA5CNFSM4KAKUD3KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEICG2KI#issuecomment-570715433",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

beikeland, 2020-01-04:
> i have done configuration from https://sdavi.github.io/LPCConfigurator/ but have issue with which files are required and how to proceed now.

you would have to also get the RRF from the same guy as he is the one who maintain the LPC port
https://github.com/sdavi/RepRapFirmware

aceranic, 2020-01-04:
Ok, thanks!!Poslano s pametnog telefona Samsung Galaxy.
-------- Izvorna poruka --------Šalje: Bjørn <notifications@github.com> Datum: 04. 01. 2020.  13:56  (GMT+01:00) To: dc42/RepRapFirmware <RepRapFirmware@noreply.github.com> Cc: aceranic <aceranic1@gmail.com>, Author <author@noreply.github.com> Predmet: Re: [dc42/RepRapFirmware] MKS Sbase v1.3 installation (#346) 
i have done configuration from https://sdavi.github.io/LPCConfigurator/ but have issue with which files are required and how to proceed now.

you would have to also get the RRF from the same guy as he is the one who maintain the LPC port
https://github.com/sdavi/RepRapFirmware

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/dc42/RepRapFirmware/issues/346?email_source=notifications\u0026email_token=AEBDTJMCJ7NSBSPAIUSG2G3Q4CBPHA5CNFSM4KAKUD3KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEICXNRA#issuecomment-570783428",
"url": "https://github.com/dc42/RepRapFirmware/issues/346?email_source=notifications\u0026email_token=AEBDTJMCJ7NSBSPAIUSG2G3Q4CBPHA5CNFSM4KAKUD3KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEICXNRA#issuecomment-570783428",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2020-10-22 by T3P3