Duet3D/RepRapFirmware #347: Z axis Configuration bug when using the RRF Config tool

JoshWobbles, 2020-01-05:
I just updated to RRF 3, and created a new config for my Ender 3 Pro, but I get "Failed to Enable Endstops" for my Z axis.

I use the original endstop switch and have no z probe installed or configured.

JoshWobbles, 2020-01-05:
I believe the issue lies in the RRF config builder, the default settings do not have a endstop pin assigned for Z 
[image]

JoshWobbles, 2020-01-05:
Now I'm getting heater faults for the bed.

Somebody needs to take a look at the default Ender 3 / RRF 3 settings because there seem to be some issues.

Billtopia, 2020-01-31:
On the CR-10s I had heat bed issues and ran the PID tuning on it and it solved the error... did you try that?

x0rtrunks, 2020-10-22:
Z endstop is properly selected now.

closed 2020-10-22 by x0rtrunks