Duet3D/RepRapFirmware #349: [RRF3] bedtemp sensor always shows -273.1°C

ikarisan, 2020-01-08:
Hi!

A few minutes ago I updated my working Duet2 Wifi 1.04 from FW 2.05 to 3.0final. Everything seems to work fine, but my bed temperature is always shown an -273.1°C. :( 

This is my configuration for the bed thermistor/heater:

M308 S0 P"bedtemp" Y"thermistor" T10000 B3950  
M950 H0 C"bedheat" T0  
M143 H0 S100
M307 H0 B0 S1.00

The hotend thermistor is working fine. 

Board: Duet WiFi 1.02 or later
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.0 (2020-01-03b3)
Duet WiFi Server Version: 1.23

StephenRC, 2020-01-09:
Your missing a zero for the T parameter.  (100000)

On Wed, 08 Jan 2020 12:23:29 -0800, Thorsten
<notifications@github.com> had a flock of green cheek conures squawk
out:

> Hi!
>
>
>
> A few minutes ago I updated my working Duet2 Wifi 1.04 from FW 2.05 to 3.0final. Everything seems to work fine, but my bed temperature is always shown an -273.1°C. :(
>
>
>
> This is my [...]

ikarisan, 2020-01-09:
Hi!
Yes, sorry. You were two minutes too fast with your answer. I was just about to delete the report again. :) 

I selected the wrong 3950 sensor in the configuration tool.
Unfortunately I did not see the missing zero.  :( 

> Your missing a zero for the T parameter. (100000)

closed 2020-01-09 by ikarisan