Duet3D/RepRapFirmware #350: [RRF2] Extrusion control in the web UI has "dangerous" defaults

orcinus, 2020-01-09:
Unsure whether this belongs in this repo's issues or some others, let me know if this needs to be submitted elsewhere.

The Extrusion Control section in Dashboard of the Duet Web Control 2.0.4 defaults to 100mm and 60 mm/s. This is potentially unsafe, as if you forget to reset them to sane settings after a reboot (e.g. after an update) and hit Extrude, you'll end up with one LONG gear-jarring skip, or chewed up filament.

Ask me how i know.

(In fact, on dual drive extruders like the Bondtech, or the new E3D Hemera, with a strong stepper, you'll likely neatly cut the filament in half, requiring opening the extruder to pull the loose piece of filament out of the bowden tube.)

It would make much more sense to have them default to safest setting, i.e. 1mm & 1mm/s.

chrishamm, 2020-01-09:
Wrong repository but anyway (see [here](https://github.com/chrishamm/DuetWebControl/issues) for the right one): The defaults in DWC2 are meant to be set to 10mm and 5mm/s but it looks like these defaults are overwritten when a connection is established. I'll have a look at it.

closed 2020-01-09 by chrishamm

orcinus, 2020-01-09:
Sorry about that, i looked around dc42 repos and only saw old DWC, so didn't know where to put it.