Duet3D/RepRapFirmware #352: [RRF3 3.01-beta3 or earlier] Filament profile issues I found using a ToolChanger

Nibbels, 2020-02-02:
Board: Duet WiFi 1.02 or later + DueX5
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.01-beta3 (2020-01-29b1)
Duet WiFi Server Version: 1.23

I use a duet2 + duex5 to drive a E3D toolchanger and want to follow the absolutly great idea of detaching the filament specific properties from my slicer. Thus I added a config in "Filaments" within the duet web UI.

1) I am still digging myself into the duet firmware and I am no expert yet. But here is some unnecessary behaviour (i think):
[image]
- Why should I not be able to assign one filament configuration to two tools?
  In my case I added a PETG configuration which can be totally valid for blue and green PETG.

2) When I switch the filament-configuration in the web UI using this linkbutton
[image]
then the printer always selects/grabs the tool if the printer is homed.
Andelse it always only raises temperatures to standby if the printer is unhomed.

But what I actually thought that would happen when choosing another filament type is that I just select the right filament for future M703 activation. No tools selection, no heatup. I want to leave the tools in state "off".
(As a toolchanger owner I want set the filament type prior to printing and prior to homing, this stratey looks very comfortable to me)

If I manually grab a tool via console T1. And then send M701 S"285 PETG" then the behaviour is what I expected. A M703 afterwards loads the temperatures. So it is a strange behaviour of the web UI I think.
Do I get the theory of use wrong here?

3) If I want to download a Profile:
[image]

[image]

[image]
load.g and unload.g are empty.
Sometimes I get a download but it only contains the empty files.
[image]

When I put some comment into load.g then the error refers to the load.g
[image]
[image]

The download of the files directly is possible.
[image]
[image]
[image]

cdedwards, 2020-05-23:
btw the JSON error when downloading a filament type, is actually an error from the web UI. This has been fixed in a very recent version of the web interface

T3P3, 2023-05-04:
If these issues persit, please open a forum thread to discuss them.

closed 2023-05-04 by T3P3