Duet3D/RepRapFirmware #353: RepRapFirmware for Duet 2 WiFi/Ethernet 3.01-RC1 OM array numerical index errors

Billtopia, 2020-02-08:
When attempting to access tool[0].active through gcode macro (or even through a locally sent echo) I get an error stating that array index is not provided although if I send the command with a '1' instead of a '0' I get an array out of bounds error as (I have no tool '1') which is what I would expect. Accessing this value through the rr_model works as expected though and gives the expected value (what the hot end active temp is set to)

Billtopia, 2020-02-20:
With 3.01 RC2 the problem still persists(only tried with an echo)

dc42, 2020-04-14:
Please re-test with 3.01-RC7.

Billtopia, 2020-04-15:
an echo test of tools[0].active / standby or the like still returns a missing index error and tools[1].xxx still comes back as out of bounds (as only 1 tool it is the expected result) but I know that the problem isn't throughout the whole OM as in my homeall.g I use
G1 X{290 + {sensors.probes[0].offsets[0] * -1}} Y{ 290 + {sensors.probes[0].offsets[1] * -1}} F6000 
to move to the back corner for homing Z back there and it computes the position of the probe just fine and there are '0's in there and the echoes on those OM keys returns the proper values

dc42, 2020-04-15:
You need to use tools[0].active[0] because you can have more than one heater on a tool. It's the second array index that is missing.

closed 2020-04-15 by dc42