Duet3D/RepRapFirmware #355: Duet3 fans do not function when mapping to heater.

adragoset, 2020-02-15:
DuetSoftwareFramework 1.2.4
RepRapFirmware v3.0

Relevant Heater Fan and tool configuration
```
; Heaters
M308 S0 P"temp0" Y"thermistor" T100000 B4138       ; configure sensor 0 as thermistor on pin temp0
M950 H0 C"out0" T0                                 ; create bed heater output on out0 and map it to sensor 0
M143 H0 S120                                       ; set temperature limit for heater 0 to 120C
M307 H0 A255.3 C291.3 D1.7 V23.6 B0                ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                            ; map heated bed to heater 0
M308 S1 P"temp1" Y"thermistor" T100000 B4138       ; configure sensor 1 as thermistor on pin temp1
M950 H1 C"out1" T1                                 ; create chamber heater output on out1 and map it to sensor 1
M143 H1 S280                                       ; set temperature limit for heater 1 to 280C
M307 H1 A30.8 C818.2 D2.8 V23.6 B0                ; disable bang-bang mode for the chamber heater and set PWM limit
M141 H1                                            ; map chamber to heater 1
M308 S2 P"temp2" Y"thermistor" T100000 B4138       ; configure sensor 2 as thermistor on pin temp2
M950 H2 C"out2" T2                                 ; create nozzle heater output on out2 and map it to sensor 2
M143 H2 S280                                       ; set temperature limit for heater 2 to 280C
M307 H2 B0 S1.00                                   ; disable bang-bang mode for heater  and set PWM limit
M308 S3 P"temp3" Y"thermistor" T100000 B4138       ; configure sensor 3 as thermistor on pin temp3
M950 H3 C"out3" T3                                 ; create nozzle heater output on 1.out0 and map it to sensor 3
M143 H3 S280                                       ; set temperature limit for heater 3 to 280C
M307 H3 A878.3 C271.9 D6.9 V23.4 B0                ; disable bang-bang mode for heater  and set PWM limit

; Fans
M950 F0 C"!out6" Q25000                            ; create fan 0 on pin !out6 and set its frequency
M106 P0 L0 C"EC +" S1 H-1                          ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"!out4" Q25000                            ; create fan 1 on pin !out5 and set its frequency
M106 P1 L0 C"EC -" S1 H-1                          ; set fan 1 value. Thermostatic control is turned off
M950 F2 C"!out5" Q25000                            ; create fan 2 on pin !out4 and set its frequency
M106 P2 L0 C"Chamber Vent" S0 H-1                  ; set fan 2 value. Thermostatic control is turned off
M950 F3 C"1.out7" Q500                             ; create fan 3 on pin out7 and set its frequency
M106 P3 S1 T100 H3                                   ; set fan 3 value. Thermostatic control is turned on                               
M950 F4 C"1.out6" Q500                             ; create fan 4 on pin out8 and set its frequency
M106 P4 S0 H-1                                     ; set fan 4 value. Thermostatic control is turned off
M950 F5 C"1.out8" Q500                             ; create fan 5 on pin out9 and set its frequency
M106 P5 S1 H-1                                     ; set fan 5 value. Thermostatic control is turned off

; Tools
M563 P0 D0 H3 F4                                    ; define tool 0
G10 P0 X0 Y0 Z0                                    ; set tool 0 axis offsets
G10 P0 R0 S0                                       ; set initial tool 0 active and standby temperatures to 0C
```
In this case Fan3 does not turn on at temperatures going over 100c.

I can manually activate the fan with :
```M106 P3 S1 H-1```
Or deactivate with: 
``` M106 P3 S0 H-1```

Temperature of that heater and tool selection work correctly and are displayed properly.

Billtopia, 2020-02-23:
Try M106 P3 S0 L1 T100 H3 
Or navigate the rr_model and find the fan and see if it says it is thermostatically controlled

Billtopia, 2020-02-24:
you can also check /rr_model?key=sensors.analog to make sure the sensor is created

T3P3, 2023-05-04:
Please use the forum for support with configuration issues

closed 2023-05-04 by T3P3