Duet3D/RepRapFirmware #358: 3.01-RC2 Stepper motors now loud in both moving and standby/idle

Billtopia, 2020-02-19:
something has changed between 3.01-RC1 and 3.01-RC2 that has caused the stepper motors to become louder while both moving and in idle state(30%). Before this update the stepper motors were quiet/ near silent while moving and silent while not in idle and not moving and now they almost make a hissing noise( or pc fan noise). I had no M569 commands previously but I saw that they may assist in quieting the steppers (https://duet3d.dozuki.com/Wiki/Reducing_Stand-Still_Noise) so I decided to try adding them. While attempting different M569 P2 commands F1 got the Z motors at 100% to about the volume of the idle state (30% power) without the M569. The commands got me nowhere near the quiet I had before 3.01-RC2

JPluess, 2020-02-20:
I was searching the web because my motors are loud too (at faster print speeds very loud). If there is any solution im interested i use the WIFI Board with firmware 3.0. 

here is my config: 

[image]

T3P3, 2020-10-20:
Please test with the latest stable version and open a forum discussion if this is still an issue

closed 2020-10-20 by T3P3