Duet3D/RepRapFirmware #362: Feature Request: Configurable speed limit for print moves with extrusion

Nibbels, 2020-02-29:
I write this request because I use duet side filament configurations.
That includes:
+ Temperature settings for idle and active state
+ hardware retraction

+ Plus it would be awsome to limit the printing feedrate for specific types or filament by putting some M20x code inside the config.g of each affected filament.

**I do not want to affect the travel feedrate.**
We normally would have a
G0 : Rapid Move
G1 : Controlled *(linear) Move
which could be used for fast travel moves or printing moves. But the documentation at https://duet3d.dozuki.com/Wiki/Gcode sais:
```
RepRapFirmware treats G0 and G1 in the same way 'except as follows:
[...]
In Laser and CNC mode, G0 moves are executed at the maximum feed rate available, to comply with the NIST GCode standard, This feed rate is set by the M203 command.
```

And I read that cura might be using G0 and G1 correctly:
```
https://3dprinting.stackexchange.com/questions/3315/why-does-cura-insert-g0-commands-between-g1-commands-while-slic3r-doesnt
```

_Some maybe hack to achieve what I want:
I suppose that we already could define the slicers travel feedrate way higher than the printers feedrate limit. https://duet3d.dozuki.com/Wiki/Gcode#Section_M203_Set_maximum_feedrate
And if I set the slicers extrusion move feedrate to some value as 100mm/s
Then we might be able to scale the feedrate for every filament using https://duet3d.dozuki.com/Wiki/Gcode#Section_M220_Set_speed_factor_override_percentage
which should not affect the travel feedrate because this is still atop of maximum. (?)
Then a M220 S80 means 80mm/s, but travel feedrate still at max. (...)_

But nevertheless I want to write this feature request because in my eyes the hardware filament configuration is the future. It makes gcode much more transferable between machines.
A gcode setting for a "hardware printing feedrate limit" would be a clean solution to this problem.

Billtopia, 2020-03-01:
Well... you can, right now, limit the maximum rate that you can extrude filament at and that would cause extrusion moves to go slower once the move tries to extrude faster than the limit...
M203 E1500.00 ; my setting
so lower the E value by half or a 2/3rd?

Nibbels, 2020-03-01:
@Billtopia 
Thank you for your reply. That might be a good solution.
Using that limit the filament is purely extruder capability dependend. Not drive system dependend.
I once saw this option but ignored it because of interaction with the retraction speed. But I switched to hardware retraction using G10 / G11.
I will test as soon as I can if the setting affects/limits the hardware rectract speeds or not.

T3P3, 2023-05-04:
Having a separate maximum federate for G0 moves would be useful, not just for extrusion machines but for CNC machines as well.

dc42, 2023-05-04:
In CNC mode, G0 moves already occur at the maximum speed configured by M203.

T3P3, 2023-05-04:
Indeed, however the OP wants to have a lower max speed, filament dependent for G1 moves. Hence splitting them.

Nibbels, 2023-09-30:
> I will test as soon as I can if the setting affects/limits the hardware rectract speeds or not.
Sorry for the late reply. I could not continue printing the last years.
About the test:

Under filaments I did set a hardware retract configuration.
M207 S1.3 F1800 Z0.2   ; Settings for hardware retract
Then I did set 
M203 to more or less than "1800", the retract feedrate.

_Test 1:_
M203 E10000
G10
G11 
-> Retract is fast. Klick-klack.

_Test 2:_
M203 E100
G10
G11 
-> Retract is slow. Brrrrrrrrrrrrrrr-brrrrrrr.

Thus the M203 appears to be a global hard limit that is limiting hardware retracts as well. That would break the retract performance if I set the feed limit per filament with M203.

My unmet goal is still:
**I want some extrusion limit, so that I can set the flow rate capability per hotend / filament.**

For example: 
I switch the filament to PETG -> max 20 mm³/s
I switch the filament to PLA -> max. 23 mm³/s
I switch the filament to Flex -> max. 8 mm³/s
(The Gcode does not know this limit and would become more filament independend.)

T3P3, 2026-08-24:
lot more info in this issue (closed as it was a duplicate):
https://github.com/Duet3D/RepRapFirmware/issues/1223

T3P3, 2026-09-16:
Forum thread: https://forum.duet3d.com/topic/36482