Duet3D/RepRapFirmware #363: Bug? Firmware 3.0 (2020-01-03b3) abort of print on "cura pause at height" feature

SchlSE, 2020-03-01:
If Cura has the pause at height feature included in the gcode, the printer will "abort" the print at this layer.
With abort is meant that the printer thinks that he finished the print ( 100% complete) while the printed layers are less then the max layers and the extruded material is also less then the max material.
There is no button, ect. that let me resume the print. 
[bug.zip](https://github.com/dc42/RepRapFirmware/files/4271552/bug.zip)
those are basically different models, but both will trigger the same behaviour

TinkerGnome, 2020-03-01:
The "Pause at height" post-processing script of Cura is designed to be used with Marlin firmware and inserts "M0" into the gcode file. The intended behavior on RRF is to cancel the print - as documented here:
[https://duet3d.dozuki.com/Wiki/Gcode#Section_M0_Stop_or_Unconditional_stop](https://duet3d.dozuki.com/Wiki/Gcode#Section_M0_Stop_or_Unconditional_stop)

Any recent version of Cura should provide a separate post-processing script for RRF, which was contributed in 2018 by Thomas Kriechbaumer (@Kriechi).
The name is **"[Pause at height for RepRapFirmware DuetWifi / Duet Ethernet / Duet Maestro](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/PauseAtHeightRepRapFirmwareDuet.py)"**

The latter inserts "M226" which triggers the pause function on RRF:
[https://duet3d.dozuki.com/Wiki/Gcode#Section_M226_Gcode_Initiated_Pause](https://duet3d.dozuki.com/Wiki/Gcode#Section_M226_Gcode_Initiated_Pause)
<br>

[image]

SchlSE, 2020-03-01:
ok thanks.

closed 2020-03-01 by SchlSE