Duet3D/RepRapFirmware #367: Wrong version number

AndWik, 2020-03-12:
Installed RepRapFirmware 3.01-RC3 - Duet2CombinedFirmware.bin
on Duet2 Wifi, from github, but it reports 2.04RC4

Board: Duet WiFi 1.02 or later
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 2.04RC4 (2019-10-19b1)
Duet WiFi Server Version: 1.23

Nibbels, 2020-03-12:
Sure?

I just downloaded this file to test it:
[image]

Then I uploaded it again and saw this:

[image]

AndWik, 2020-03-13:
I created a new directory and downloaded the file again. Used "Upload system files" and accepted install. The Machine Specific screen still tells me I'm using RRF 2.04.
I load the bin-file from the network over WiFi but this has never caused any problem before.
Is there something I have missed?

M122
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 2.04RC4 running on Duet WiFi 1.02 or later
Board ID: 08DGM-917NK-F23T0-6JKD0-3S86J-TYBJF
Used output buffers: 3 of 24 (8 max)
=== RTOS ===
Static ram: 25700
Dynamic ram: 92748 of which 0 recycled
Exception stack ram used: 268
Never used ram: 12356
Tasks: NETWORK(ready,628) HEAT(blocked,1232) MAIN(running,3808) IDLE(ready,160)
Owned mutexes:
=== Platform ===
Last reset 00:05:14 ago, cause: software
Last software reset time unknown, reason: User, spinning module GCodes, available RAM 11360 bytes (slot 3)
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0441f000 BFAR 0xe000ed38 SP 0xffffffff Task 0x4e49414d
Error status: 0
Free file entries: 10
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest block write time: 0.0ms, max retries 0
MCU temperature: min 27.9, current 31.1, max 31.2
Supply voltage: min 11.9, current 12.0, max 12.1, under voltage events: 0, over voltage events: 0, power good: yes
Driver 0: standstill, SG min/max not available
Driver 1: standstill, SG min/max not available
Driver 2: standstill, SG min/max not available
Driver 3: standstill, SG min/max not available
Driver 4: standstill, SG min/max not available
Date/time: 2020-03-13 07:29:36
Cache data hit count 1172500271
Slowest loop: 3.62ms; fastest: 0.06ms
I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0
=== Move ===
Hiccups: 0, FreeDm: 160, MinFreeDm: 160, MaxWait: 0ms
Bed compensation in use: none, comp offset 0.000
=== DDARing ===
Scheduled moves: 0, completed moves: 0, StepErrors: 0, LaErrors: 0, Underruns: 0, 0
=== Heat ===
Bed heaters = 0 -1 -1 -1, chamberHeaters = -1 -1
=== GCodes ===
Segments left: 0
Stack records: 1 allocated, 0 in use
Movement lock held by null
http is idle in state(s) 0
telnet is idle in state(s) 0
file is idle in state(s) 0
serial is idle in state(s) 0
aux is idle in state(s) 0
daemon is idle in state(s) 0
queue is idle in state(s) 0
autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 39.90ms; fastest: 0.00ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0) Telnet(0)
HTTP sessions: 1 of 8
- WiFi -
Network state is running
WiFi module is connected to access point 
Failed messages: pending 0, notready 0, noresp 0
WiFi firmware version 1.23
WiFi MAC address b4:e6:2d:53:16:24
WiFi Vcc 3.40, reset reason Turned on by main processor
WiFi flash size 4194304, free heap 24752
WiFi IP address 192.168.xxx.xxx
WiFi signal strength -46dBm, reconnections 0, sleep mode modem
Socket states: 0 0 0 0 0 0 0 0

Billtopia, 2020-03-13:
Did you upgrade to v3.0 first? https://github.com/dc42/RepRapFirmware/releases/tag/3.0
AFAIK there is no direct upgrade to latest from v2.x. You need to download the full zip from above, install it, then upgrade to the latest as in the whats_new.md as some files were updated in v3.0 that are required to be updated in order to allow the firmware to be updated past v3.0.

Billtopia, 2020-03-13:
You will also have to check your config files to make sure that you update the g-codes as command options have changed on some of them

AndWik, 2020-03-13:
Thanks a lot! Maybe I'm stupid but it was not obvious and not written in the instructions.

Many, many thanks
Anders
Sweden

closed 2020-04-14 by dc42