Duet3D/RepRapFirmware #370: Support Dwell (G4) in Laser mode.

LarsGBV, 2020-03-16:
Hi, I have build a diode laser cnc and I am having a little issue which I think 
is due to the diode having a small warm up time. 

So I would like to call a dwell with the laser on for 100ms but it seems to be impossible with the instruction set.

Is it possible somehow or might it be introduced at some point ?

My workaround for the moment is a short movement on the Z axis in my post processor
to simulate a dwell. A very short XY axis movement would also be possible I guess.

I would also be nice for when you want to turn on the laser, on low power, to focus it or similar,

T3P3, 2023-05-04:
Please comment if this is still a requirement, how common are lasers that require a dwell before they come up to power an motion can start?

For the focus request there is another issues raised:
https://github.com/Duet3D/RepRapFirmware/issues/697

ghawkesford, 2023-05-04:
Seems more than reasonable to me that G4 should be available in laser mode, can't understand why it should have been taken out, I use it in CNC mode for a similar thing when starting the spindle.

ghawkesford, 2023-05-04:
That's strange then, from LarsGBVs comments, but I do recall from the dim and distant past that there are things to do at the start of a laser cut, possibly more than just a dwell??  In any case, I would imagine that it's a post-processor issue.

T3P3, 2023-05-04:
I think the comment makes sense as he is commanding a Z move as a work around (so start firing the laser, do a short Z move then start the desired motion). Anyway we need owners of these sorts of lasers to confirm the requirement.

dc42, 2023-05-04:
I am not aware of any other controller firmware that supports this. Perhaps it can it be simulated by reducing acceleration?

T3P3, 2024-03-06:
please reopen if still relevant

closed 2024-03-06 by T3P3