Duet3D/RepRapFirmware #371: 3.01-RC4 not installable (Duet2CombinedAP.bin not found)

bernhardberger, 2020-03-17:
I just tried to install the newest Duet2Combined.bin for 3.01-RC4 on my Duet Wifi. However it says `M997 S0 - Error: In-Application programming binary Duet2Combined**AP**.bin not found.

[image]

Billtopia, 2020-03-17:
Did you upgrade to v3.0 first? https://github.com/dc42/RepRapFirmware/releases/tag/3.0
AFAIK there is no direct upgrade to latest from v2.x. You need to download the full zip from above, install it, then upgrade to the latest as in the whats_new.md as some files were updated in v3.0 that are required to be updated in order to allow the firmware to be updated past v3.0.

bernhardberger, 2020-03-17:
Sorry, I forgot to mention that: I'm trying to update from 3.01-RC3

bernhardberger, 2020-03-17:
Nevermind. Just found the missing Duet2Combined IAP binary in the zip archive from the 3.0 release linked above.

Can be closed.

However: I wished the necessary assets would be linked in the releases. Upgrading FW is often kind of an adventure searching through a ton of previous releases for the required files, and in some cases even different repositories :/

closed 2020-03-17 by bernhardberger